Klipper3d/klipper #1499: SKR/LPC1768 USB hangs on FIRMWARE_RESET

mattthebaker, 2019-04-04:
Board becomes unresponsive to USB after FIRMWARE_RESET. Confirmed to affect at least SKR 1.1/1.3, and possibly other LPC176x models.

It is somewhat intermittent (can take from 1-10 resets), and seems to depend on operating conditions (during board bring up every reset caused this, but it is now less frequent).

After testing power, lcd, and stepstick configurations to rule them out, it appears to be the same root cause as issue #1470 .  The host doesn't recognize a USB disconnect event and the USB hangs.

Here is the dmesg log from a proper reset, followed by a failed reset, attempts to do an arduino style reset by klipper, and at the end a pulled usb cable. Note that the final disconnect device number matches the device number in the preceding reset.

```
[40484.459175] usb 1-1.5: USB disconnect, device number 31
[40499.086564] usb 1-1.5: new full-speed USB device number 32 using dwc_otg
[40499.220772] usb 1-1.5: New USB device found, idVendor=2341, idProduct=abcd
[40499.220786] usb 1-1.5: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[40499.220795] usb 1-1.5: Product: Klipper firmware
[40499.220803] usb 1-1.5: Manufacturer: Klipper
[40499.220811] usb 1-1.5: SerialNumber: 12345
[40499.221779] cdc_acm 1-1.5:1.0: ttyACM1: USB ACM device

[40517.581759] cdc_acm 1-1.5:1.0: failed to set dtr/rts
[40522.610505] cdc_acm 1-1.5:1.0: failed to set dtr/rts
[40527.636129] cdc_acm 1-1.5:1.0: failed to set dtr/rts
[40532.664875] cdc_acm 1-1.5:1.0: failed to set dtr/rts
[40533.355192] usb 1-1.5: USB disconnect, device number 32
```

I ultimately tested the #1470 solution by putting a similar delay in the reset command call, and separately before starting usb serial. Both cases eliminated the hang. Shorter delays (<1ms) did not seem to be effective. I did not put a pull request together because both solutions seemed like hacks.

For what its worth, I checked the USB2 spec, and the disconnect time (TDDIS) when disabling the D+ pullup is supposed to be 2.5us. Delays that short were not effective though. There is a concept of "reset protocol" in usb that may be complicating the detection of disconnects and has spec numbers in the range of 3-7ms. My understanding of USB protocol is very shallow.

I also estimated that if a chip reset is issued, instead of USB_CONN being actively driven high before reset, it would require about 50us for the weak pull up in the io buffer to completely pull the pin high.

It may make the most sense to have the delay before initiating the usb connection, as it would cover reset cases other than by command (power glitches, reset switch).  If the failed dtr/rts is communicated by error code, it might be possible to resolve from the host side (some boards have alternative 5v sources, but even the host resetting state should probably recover it).

[klippy.log](https://github.com/KevinOConnor/klipper/files/3044568/klippy.log)

Iakabos, 2019-04-04:
I'm also having this problem. For me, with dual SKR V1.3s, it happens every time I try FIRMWARE_RESTART. If I disconnect the 5v supply (either unplugging USB if 5v is set to USB or turning off 12v power supply if 5v is set to internal regulator) it is able to finish restarting the boards.

[klippy.log](https://github.com/KevinOConnor/klipper/files/3045843/klippy.log)

mattthebaker, 2019-04-05:
I pushed one of the experiments that adds the delay before enabling USB for reproduction. I have a feeling this isn't the best solution but will hopefully confirm the fix/issue.

https://github.com/mattthebaker/klipper/tree/lpc176x_usb_start_delay
I'm not sure if there is a better way to link to the commit...

Iakabos, 2019-04-05:
I can confirm that that commit works for me

KevinOConnor, 2019-04-06:
@mattthebaker - Thanks.  That's how I'd do it as well.  However, the code can't assume timer_read_time() works as it may not have been setup yet - take a look at the udelay() implementation in stm32f1 for an alternate mechanism.  Also, it shouldn't require an explicit call to `sie_cmd_write(SIE_CMD_SET_DEVICE_STATUS, 0);` as it should already be in an off mode - I think just the udelay(5000) would be needed.

-Kevin

mattthebaker, 2019-04-07:
@KevinOConnor Thanks for the feedback.  Agree with you on the device status -- it doesn't look like re-initialization is a thing in klipper and the state will be reset whenever there is a power on/chip reset. The init system looks very interesting. I added the method used in stm32f1 to the lpc176x code.  Maybe long term that could be moved into the arm timer code if many platforms need it.

The solution still works for me -- as far as I can tell with 20 or so restart cycles. It would be ideal to get confirmation from someone that had a higher incidence rate.

The delay code includes clock speed, so this should be equally effective on the lpc1769.  I see that PLL/Clock initialization is handled by the lpc libs before main is called, so the delay should be accurate.

In a separate commit in the pull request I also changed the issue #1470 delay to use the udelay helper.

Thanks.

KevinOConnor, 2019-04-07:
Thanks.  I went ahead and committed it - it's been an open question if a delay is needed on a few different boards (for example stm32f1 uses udelay() in its usb bringup).  I'm not sure why my dev board doesn't show the issue.  Could be due to host usb implementation.

I agree - udelay() should be moved to armcm-timer.c and a new armcm-timer.h file should be created to export it.

-Kevin

mattthebaker, 2019-04-07:
Great.

When I first setup this board, it was happening every reset. A week or two later it stopped -- I thought it had completely until someone mentioned it again and I tried repeat resets.  My first thought was temperature dependence -- winter finally turned and we quit running our heaters.

I still think it may have to do with RC delays, temperature, and the D+ line.  2.5us of D+ low while the bus is idle should trigger a disconnect. I did not get the scope out to look though.

It also really sticks out in my mind that big tree used a **5A** mosfet to switch the 1.5k pullup on D+.  That's why I thought the chip reset default pin state might not disable it quick enough.

Actually maybe that thing just has so much active area it leaks too much current onto D+. Worst case leakage is 150uA at -16V Vds, 125C. The pull down in the hub would be sinking only 220uA at 3.3v, and less at lower voltages. Its within an order of magnitude.

Would explain why not all designs exhibit it, and could be the same cause on other non lpc designs. I don't think I saw the issue on my sbase but I wasn't looking either.  Chips with the pull up integrated would be more consistent.

closed 2019-04-07 by mattthebaker