Klipper3d/klipper #1500: Problems connecting to printer

blamedcash, 2019-04-05:
I installed Klipper and have been using it for several weeks now. It’s great; however, the one challenge I’ve had is getting the printer to connect without going through the original connection process (creating a /tmp/printer connection, etc). I followed all of the directions to the tee and in fact, flashed it twice with the same results. The only way I can use octoprint and Klipper is through the following steps after a restart of the pi. BTW - restarting octoprint alone works fine - this only happens when I restart the pi. I have the correct path added for serial to my config file too, but to no avail. I’ve read all of the FAQ too.

1. Delete the /tmp/printer
2. Try logging on to /dev/ttyUSB0 (fails to connect, but I can’t skip this step)
3. Then follow the original directions (add /tmp/printer, save, set to /tmp/printer, save, set connection to /tmp/printer, and then connect).

The above process works. Otherwise I get a terminal message: “Printer is not ready”. 
My printer is a Wanhao Duplicator I3 Plus.
[klippy.log](https://github.com/KevinOConnor/klipper/files/3047562/klippy.log)

KevinOConnor, 2019-04-07:
It sounds like something in the OctoPrint installation is corrupted.  The Klipper log doesn't provide any further information.  I'd say backup what you need and then do a new sd card installation of OctoPi.

-Kevin

blamedcash, 2019-04-07:
That was it! Thank you sir.

Tom

blamedcash, 2019-04-07:
Problem solved

closed 2019-04-07 by blamedcash