Klipper3d/klipper #1501: ImportError: No module named jinja2 after update

noud-github, 2019-04-05:
Did a git pull today, updated mcu firmware,
looks like i encountered missing module..... 
doing a successful apt-get install python-jinja2, did not help 

=======================
Unhanded exception during connect
Trackback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 133, in _connect
    self._read_config()
  File "/home/pi/klipper/klippy/klippy.py", line 126, in _read_config
    self.try_load_module(config, section_config.get_name())
  File "/home/pi/klipper/klippy/klippy.py", line 109, in try_load_module
    mod = importlib.import_module('extras.' + module_name)
  File "/usr/lib/python2.7/importlib/__init__.py", line 37, in import_module
    __import__(name)
  File "/home/pi/klipper/klippy/extras/gcode_macro.py", line 7, in <module>
    import jinja2
ImportError: No module named jinja2

[klippy.log](https://github.com/KevinOConnor/klipper/files/3049155/klippy.log)

Arksine, 2019-04-05:
You can run the install script to install/update the new dependencies.  For most people this will be scripts/install_octopi.sh

noud-github, 2019-04-05:
Thanx that fixed it

closed 2019-04-05 by noud-github

LazaroFilm, 2019-04-20:
Do you have a step by step to do that?

KevinOConnor, 2019-04-20:
https://www.klipper3d.org/FAQ.html#how-do-i-upgrade-to-the-latest-software

-Kevin

supernielsen, 2019-06-19:
Hi everybody i have this issue aswell and install_octopi.sh dont work here.. What to do?

This happened directly after update aswell.

YaronSoffer, 2019-08-22:
connect through the  terminal and type the follow command 
"sudo pip install jinja2"

This solved it in my case - the update did not

YaronSoffer, 2019-08-22:
In my second machine what worked was :
cd ~/klippy-env/bin
 sudo ./pip install jinja2