Klipper3d/klipper #1502: Steppers motors wrong direction

Ernata, 2019-04-05:
[klippy.log](https://github.com/KevinOConnor/klipper/files/3049692/klippy.log)
Hello,

I have an issue with the stepper motors direction on all axis. When I press home all the motors are turning in the good direction and detect properly the end stops. But when I ask the motors to move they stil try to go on the end stop direction. I tried to change the direction of step_pin but putting a ! or not is changing anything.
What I'm doing wrong ? 
Thanks in advance, best regards,
Serge

jakep82, 2019-04-05:
The direction pin is dir_pin, not step_pin.  You should not reverse the step_pin.

Ernata, 2019-04-06:
I already tried to change the direction on dir_pin. The steppers motors are turning on the opposite direction so when I press the "Home fonction" it's moving away from the end stops. And even in this case if I manualy press the end stops and trie to move the carriage it still go go in the same direction as the "home fonction".

Ernata, 2019-04-06:
I found the mistake. I changed the end stop logic (with a ! ) and know everything is working properly.

Thanks

closed 2019-04-06 by Ernata