Klipper3d/klipper #1507: pid_deriv_time error

MrMbss1, 2019-04-06:
I'm having issues getting my printer to work after attempting to update the Klipper firmware. I followed the FAQ guide on how to update it. After the update, it's giving me this error: "Option 'pid_deriv_time' is not valid in section 'heater_bed' Once the underlying issue is corrected, use the "RESTART" command to reload the config and restart the host software. Printer is halted". I've tried changing the value between 0 and 10 in the config, but the issue persists.

What am I doing wrong?

When I look through the directories through my SFTP client, it looks like there's 2 different installs of Klipper inside one another, but I may be wrong. See attached screenshot. 

[klippy.log](https://github.com/KevinOConnor/klipper/files/3050436/klippy.log)

[image]

[image]

Hywelmartin, 2019-04-06:
Have you tried to remove the pid_deriv_time.

For all the valid settings have a look in example.cfg

MrMbss1, 2019-04-06:
@Hywelmartin
It wasn't even included in the config file to begin with, I added it to try and change the value to see if it fixed it. Sorry for not including that in the main post.

I've tried setting it at 2, which is the default value in example.cfg, no change

Hywelmartin, 2019-04-06:
But that setting is for [Temperature-triggered cooling fans](https://github.com/KevinOConnor/klipper/blob/8c54fc8753031cea6366901538dcb94ef3de5b8e/config/example-extras.cfg#L593)
You are maybe looking for[smooth_time: 2.0](https://github.com/KevinOConnor/klipper/blob/8c54fc8753031cea6366901538dcb94ef3de5b8e/config/example.cfg#L170)

MrMbss1, 2019-04-06:
I'm not sure I understand what you mean, can you elaborate?

I've completely removed the pid_deriv_time from the heatbed section now, as it was originally, yet it still gives me the same error. See attached 
[image]
[image]

ShohninDmitriy, 2019-04-06:
[heater_bed]
heater_pin: P2.5
max_power: 1.0
sensor_type: NTC 100K MGB18-104F39050L32
sensor_pin: P0.23
control: pid
pid_Kp: 107.845
pid_Ki: 4.085
pid_Kd: 711.780
**pwm_cycle_time: 0.035**  Here is this string forgotten
min_temp: 0
max_temp: 130

MrMbss1, 2019-04-06:
@ShohninDmitriy
I don't recall ever including that string, but I may be wrong. I'll try including it when I get back home sometime tomorrow

Hywelmartin, 2019-04-06:
When you updated, did you flash your controller board(s) as well?

MrMbss1, 2019-04-06:
@Hywelmartin
Yes, I flashed the board after updating. Also yes, I've also tried restarting OctoPrint and reboot the entire system multiple times

jakep82, 2019-04-06:
Are you sure you're editing the correct config file?

MrMbss1, 2019-04-06:
@jakep82
I'm editing the one in pi/home. Should be the correct one, yes?

jakep82, 2019-04-06:
If you removed the line and you're still getting the error, then you edited the wrong file. SSH into your pi and type `nano ~/printer.cfg`

See if the pid_deriv_time line is still there.

MrMbss1, 2019-04-06:
@jakep82
That's exactly what I've been doing. The line is not there

KevinOConnor, 2019-04-07:
It looks like you've got a corrupt installation somehow.  You can try backing up your printer.cfg file, running the `~/klipper/scripts/klipper-uninstall.sh` script, manually running the `rm` command that script reports, and then rerunning the full installation steps from the Installation document.  If that doesn't help, try a complete reinstall of the sd card with OctoPi.

Be aware that config sections in printer.cfg are not required to be together, so if there is something odd in the config file, you'll need to check the entire file for it.

-Kevin

MrMbss1, 2019-04-07:
@KevinOConnor 
I did as you said, and now it works. A thousand times thank you.

KevinOConnor, 2019-04-14:
Thanks.  I'm going to close this issue as it looks like the original issue was resolved.

-Kevin

closed 2019-04-14 by KevinOConnor