Klipper3d/klipper #1508: [FR] Silent command

Hywelmartin, 2019-04-06:
So I have been using a script to alter M204 SX to SET_VELOCITY_LIMIT ACCEL=X ACCEL_TO_DECEL=(X*Y)
#1294 

This works good until I get a filament outage.
Then Octoprint get timeouts and stops monitoring, the connection gets reconnected after some time >10min.
#1467 

Would it be possible to make this command(or other as well) silent, I normally print via v-SD.

Like
SET_VELOCITY_LIMIT ACCEL=X ACCEL_TO_DECEL=(X*Y) SILENT

[klippy.zip](https://github.com/KevinOConnor/klipper/files/3050438/klippy.zip)

KevinOConnor, 2019-04-14:
FWIW, I suspect tracking down the OctoPrint issue would be preferable to making changes to Klipper output.  Also, it should not be necessary to customize ACCEL_TO_DECEL - I recommend setting a reasonable rate (that reduces overall printer vibrations) for it - if one uses a lower overall acceleration then it is not necessary to also reduce ACCEL_TO_DECEL.

-Kevin

Hywelmartin, 2019-04-15:
this is a very repeatable behaviour, so it might be easy to track down,  but the octoprint devs are not Klipper users... 

I can try with Gina & co

closed 2019-04-15 by Hywelmartin