Klipper3d/klipper #1509: SKR v1.3 and SPI TMC2130

dzao, 2019-04-06:
Hello a I have two skr v1.3 and on both board a want to use tmc2130 in spi mode. Everything looks okay, but motors doesn´t move. If I put commad DUMP_TMC stepper=stepper_x I get:

Send: DUMP_TMC stepper=stepper_x
Recv: // GCONF:          00000000
Recv: // GSTAT:          00000000
Recv: // IOIN:           00000000
Recv: // TSTEP:          00000000
Recv: // XDIRECT:        00000000
Recv: // MSCNT:          00000000
Recv: // MSCURACT:       00000000
Recv: // CHOPCONF:       00000000
Recv: // DRV_STATUS:     00000000
Recv: // PWM_SCALE:      00000000
Recv: // LOST_STEPS:     00000000

Is it okay? Maybe i must modify out/spi.c for other SPI, but I don´t know how.

dzao, 2019-04-06:
here is a log:
Starting Klippy...
Args: ['/home/pi/klipper/klippy/klippy.py', '/home/pi/printer.cfg', '-l', '/tmp/klippy.log']
Git version: 'v0.6.0-663-g6df7356-dirty'
CPU: 4 core ARMv7 Processor rev 4 (v7l)
Python: '2.7.13 (default, Nov 24 2017, 17:33:09) \n[GCC 6.3.0 20170516]'
Start printer at Sat Apr  6 16:17:07 2019 (1554567427.1 8.6)
===== Config file =====
[mcu]
serial = /dev/serial/by-path/platform-3f980000.usb-usb-0:1.1.3:1.0

[mcu zboard]
serial = /dev/serial/by-path/platform-3f980000.usb-usb-0:1.2:1.0

[printer]
kinematics = cartesian
max_velocity = 500
max_accel = 3000
max_z_velocity = 25
max_z_accel = 30

[stepper_x]
step_pin = P2.2
dir_pin = !P2.6
enable_pin = !P2.1
step_distance = .0016
endstop_pin = !P1.29
position_endstop = 0
position_max = 300
homing_speed = 10

[tmc2130 stepper_x]
cs_pin = P1.17
microsteps = 16
interpolate = False
run_current = 1.2
hold_current = 1
sense_resistor = 0.220
stealthchop_threshold = 70

[stepper_y]
step_pin = P0.19
dir_pin = P0.20
enable_pin = !P2.8
step_distance = .0016
endstop_pin = !P1.27
position_endstop = 0
position_max = 300
homing_speed = 10

[tmc2130 stepper_y]
cs_pin = P1.15
microsteps = 128
interpolate = False
run_current = 1.2
hold_current = 1
sense_resistor = 0.220
stealthchop_threshold = 70

[stepper_z]
step_pin = zboard:P2.2
dir_pin = zboard:P2.6
enable_pin = !zboard:P2.1
step_distance = 0.0003125
endstop_pin = probe: z_virtual_endstop
position_max = 300
homing_speed = 5

[tmc2130 stepper_z]
cs_pin = zboard:P1.17
[... 168 lines cut ...]
Stats 47.6: gcodein=0 mcu: mcu_awake=0.001 mcu_task_avg=0.000006 mcu_task_stddev=0.000006 bytes_write=1531 bytes_read=7769 bytes_retransmit=9 bytes_invalid=0 send_seq=179 receive_seq=179 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=25000524 zboard: mcu_awake=0.000 mcu_task_avg=0.000005 mcu_task_stddev=0.000005 bytes_write=1403 bytes_read=3980 bytes_retransmit=9 bytes_invalid=0 send_seq=170 receive_seq=170 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=25000550 adj=25000084 heater_bed: target=0 temp=21.0 pwm=0.000 print_time=12.033 buffer_time=0.000 print_stall=0 extruder: target=0 temp=21.1 pwm=0.000 extruder: target=0 temp=21.1 pwm=0.000
Stats 48.6: gcodein=0 mcu: mcu_awake=0.001 mcu_task_avg=0.000006 mcu_task_stddev=0.000006 bytes_write=1537 bytes_read=7875 bytes_retransmit=9 bytes_invalid=0 send_seq=180 receive_seq=180 retransmit_seq=2 srtt=0.001 rttvar=0.001 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=25000521 zboard: mcu_awake=0.000 mcu_task_avg=0.000005 mcu_task_stddev=0.000005 bytes_write=1409 bytes_read=3996 bytes_retransmit=9 bytes_invalid=0 send_seq=171 receive_seq=171 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=25000551 adj=25000060 heater_bed: target=0 temp=21.0 pwm=0.000 print_time=12.033 buffer_time=0.000 print_stall=0 extruder: target=0 temp=21.1 pwm=0.000 extruder: target=0 temp=21.1 pwm=0.000
Stats 49.6: gcodein=0 mcu: mcu_awake=0.001 mcu_task_avg=0.000006 mcu_task_stddev=0.000006 bytes_write=1543 bytes_read=7996 bytes_retransmit=9 bytes_invalid=0 send_seq=181 receive_seq=181 retransmit_seq=2 srtt=0.001 rttvar=0.001 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=25000522 zboard: mcu_awake=0.000 mcu_task_avg=0.000005 mcu_task_stddev=0.000005 bytes_write=1415 bytes_read=4012 bytes_retransmit=9 bytes_invalid=0 send_seq=172 receive_seq=172 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=25000552 adj=25000079 heater_bed: target=0 temp=21.0 pwm=0.000 print_time=12.033 buffer_time=0.000 print_stall=0 extruder: target=0 temp=21.1 pwm=0.000 extruder: target=0 temp=21.1 pwm=0.000
Stats 50.6: gcodein=0 mcu: mcu_awake=0.001 mcu_task_avg=0.000006 mcu_task_stddev=0.000006 bytes_write=1549 bytes_read=8131 bytes_retransmit=9 bytes_invalid=0 send_seq=182 receive_seq=182 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=25000524 zboard: mcu_awake=0.000 mcu_task_avg=0.000005 mcu_task_stddev=0.000005 bytes_write=1421 bytes_read=4039 bytes_retransmit=9 bytes_invalid=0 send_seq=173 receive_seq=173 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=25000553 adj=25000067 heater_bed: target=0 temp=21.0 pwm=0.000 print_time=12.033 buffer_time=0.000 print_stall=0 extruder: target=0 temp=21.1 pwm=0.000 extruder: target=0 temp=21.1 pwm=0.000
Stats 51.6: gcodein=0 mcu: mcu_awake=0.001 mcu_task_avg=0.000006 mcu_task_stddev=0.000006 bytes_write=1555 bytes_read=8237 bytes_retransmit=9 bytes_invalid=0 send_seq=183 receive_seq=183 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=25000524 zboard: mcu_awake=0.000 mcu_task_avg=0.000005 mcu_task_stddev=0.000005 bytes_write=1427 bytes_read=4055 bytes_retransmit=9 bytes_invalid=0 send_seq=174 receive_seq=174 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=25000554 adj=25000053 heater_bed: target=0 temp=21.0 pwm=0.000 print_time=12.033 buffer_time=0.000 print_stall=0 extruder: target=0 temp=21.1 pwm=0.000 extruder: target=0 temp=21.1 pwm=0.000
Stats 52.6: gcodein=0 mcu: mcu_awake=0.001 mcu_task_avg=0.000006 mcu_task_stddev=0.000006 bytes_write=1561 bytes_read=8358 bytes_retransmit=9 bytes_invalid=0 send_seq=184 receive_seq=184 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=25000522 zboard: mcu_awake=0.000 mcu_task_avg=0.000005 mcu_task_stdde [... cut ...] 0 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=25000519 zboard: mcu_awake=0.000 mcu_task_avg=0.000005 mcu_task_stddev=0.000005 bytes_write=1553 bytes_read=4435 bytes_retransmit=9 bytes_invalid=0 send_seq=195 receive_seq=195 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=25000556 adj=25000038 heater_bed: target=0 temp=20.9 pwm=0.000 print_time=12.033 buffer_time=0.000 print_stall=0 extruder: target=0 temp=21.1 pwm=0.000 extruder: target=0 temp=21.1 pwm=0.000

mattthebaker, 2019-04-06:
There are config options to set the SPI pins MISO, MOSI, and SCK. Did you try them?

dzao, 2019-04-06:
Do you mean #spi_software_sclk_pin: P0.4
#spi_software_mosi_pin: P4.28
#spi_software_miso_pin: P0.5 ??

yes, i did it

Iakabos, 2019-04-07:
Please tell me you uncommented them?

dzao, 2019-04-07:
yes i know, but this is for software spi pins and doesn´t work for me

Stephan3, 2019-04-07:
With SKR 1.3 and TMC5160 i ecxperienced that the jumper for chip power has to be on internal 5v position. UART worked with both but spi only with internal 5v. Hope this helps.

dzao, 2019-04-07:
I tryed both options

dzao, 2019-04-07:
I think problem is in the spi bus, i see other pins configured in out/spi.c for spi but i dont know, how to change it. Kevin, can you check this? On SKR v1.3 are used pins: P0.5, P4.28, P0.4 (MISO,MOSI,SCK).Thank you

Stephan3, 2019-04-07:
````
[tmc5160 stepper_x]
cs_pin: P1.17
spi_software_sclk_pin: P0.4
spi_software_mosi_pin: P4.28
spi_software_miso_pin: P0.5
```
This works for spi communication on sepper x. You set the jumpers right @dzao ?

dzao, 2019-04-07:
If I use this setting a I get error: Option 'spi_software_sclk_pin' is not valid in section 'tmc2130 stepper_x'

KevinOConnor, 2019-04-07:
@dzao - you'll need to upgrade to the latest version of Klipper (see the FAQ for updating).  If you continue to have issues, please attach a new Klipper log file from that version of the code.  In the future, please be sure to _attach_ the Klipper log file.

-Kevin

dzao, 2019-04-07:
Okay, thank you this is a good idea, but now I have another problem. If i run "make" I get this error:

Compiling out/src/lpc176x/spi.o
src/lpc176x/spi.c:13:18: error: expected declaration specifiers or '...' before string constant
 DECL_ENUMERATION("spi_bus", "ssp0", 0);
                  ^
src/lpc176x/spi.c:13:29: error: expected declaration specifiers or '...' before string constant
 DECL_ENUMERATION("spi_bus", "ssp0", 0);
                             ^
src/lpc176x/spi.c:13:37: error: expected declaration specifiers or '...' before numeric constant
 DECL_ENUMERATION("spi_bus", "ssp0", 0);
                                     ^
src/lpc176x/spi.c:14:19: error: expected declaration specifiers or '...' before string constant
 DECL_CONSTANT_STR("BUS_PINS_ssp0", "P0.17,P0.18,P0.15");
                   ^
src/lpc176x/spi.c:14:36: error: expected declaration specifiers or '...' before string constant
 DECL_CONSTANT_STR("BUS_PINS_ssp0", "P0.17,P0.18,P0.15");
                                    ^
src/lpc176x/spi.c: In function 'spi_init':
src/lpc176x/spi.c:25:21: warning: implicit declaration of function 'GPIO' [-Wimplicit-function-declaration]
     gpio_peripheral(GPIO(0, 17), 2, 0);
                     ^
src/lpc176x/spi.c:25:5: error: too few arguments to function 'gpio_peripheral'
     gpio_peripheral(GPIO(0, 17), 2, 0);
     ^
In file included from src/lpc176x/spi.c:10:0:
src/lpc176x/internal.h:11:6: note: declared here
 void gpio_peripheral(int bank, int pin, int func, int pullup);
      ^
src/lpc176x/spi.c:26:5: error: too few arguments to function 'gpio_peripheral'
     gpio_peripheral(GPIO(0, 18), 2, 0);
     ^
In file included from src/lpc176x/spi.c:10:0:
src/lpc176x/internal.h:11:6: note: declared here
 void gpio_peripheral(int bank, int pin, int func, int pullup);
      ^
src/lpc176x/spi.c:27:5: error: too few arguments to function 'gpio_peripheral'
     gpio_peripheral(GPIO(0, 15), 2, 0);
     ^
In file included from src/lpc176x/spi.c:10:0:
src/lpc176x/internal.h:11:6: note: declared here
 void gpio_peripheral(int bank, int pin, int func, int pullup);
      ^
src/lpc176x/spi.c:30:5: warning: implicit declaration of function 'enable_pcloc ' [-Wimplicit-function-declaration]
     enable_pclock(PCLK_SSP0);
     ^
Makefile:64: recipe for target 'out/src/lpc176x/spi.o' failed
make: *** [out/src/lpc176x/spi.o] Error 1

GeckoSys3D, 2019-04-07:
I have tried with the SW SPI pins. 
The result seems like this:
Send: DUMP_TMC STEPPER=stepper_x
Recv: // ========== Write-only registers ==========
Recv: // IHOLD_IRUN: 00081a1f IHOLD=31 IRUN=26 IHOLDDELAY=8
Recv: // TPWMTHRS:   00000000
Recv: // TPOWERDOWN: 00000000
Recv: // PWMCONF:    00050480 PWM_AMPL=128 PWM_GRAD=4 pwm_freq=1 pwm_autoscale=1
Recv: // COOLCONF:   00000000
Recv: // ========== Queried registers ==========
Recv: // GCONF:      ffffffff I_scale_analog=1(ExtVREF) internal_Rsense=1 en_pwm_mode=1 enc_commutation=1 shaft=1(Reverse) diag0_error=1 diag0_otpw=1 diag0_stall=1 diag1_stall=1 diag1_index=1 diag1_onstate=1 diag1_steps_skipped=1 diag0_int_pushpull=1 diag1_pushpull=1 small_hysteresis=1 stop_enable=1 direct_mode=1 test_mode=1
Recv: // GSTAT:      ffffffff reset=1 drv_err=1(ErrorShutdown!) uv_cp=1(Undervoltage!)
Recv: // IOIN:       ffffffff STEP=1 DIR=1 DCEN_CFG4=1 DCIN_CFG5=1 DRV_ENN_CFG6=1 DCO=1 VERSION=0xff
Recv: // TSTEP:      ffffffff TSTEP=1048575
Recv: // XDIRECT:    ffffffff
Recv: // MSCNT:      ffffffff MSCNT=1023
Recv: // MSCURACT:   ffffffff CUR_A=-1 CUR_B=-1
Recv: // CHOPCONF:   ffffffff toff=15 hstrt=7 hend=15 fd3=1 disfdcc=1 rndtf=1 chm=1 TBL=3 vsense=1 vhighfs=1 vhighchm=1 sync=15 MRES=15(0usteps) intpol=1 dedge=1 diss2g=1
Recv: // DRV_STATUS: ffffffff SG_RESULT=1023 fsactive=1 CS_ACTUAL=31 stallGuard=1 ot=1(OvertempError!) otpw=1(OvertempWarning!) s2ga=1(ShortToGND_A!) s2gb=1(ShortToGND_B!) ola=1(OpenLoad_A!) olb=1(OpenLoad_B!) stst=1
Recv: // PWM_SCALE:  ffffffff PWM_SCALE=255
Recv: // LOST_STEPS: ffffffff LOST_STEPS=1048575
Recv: ok

looks like everything is maxed out...

KevinOConnor, 2019-04-08:
@dzao - it sounds like you skipped some of the steps in the FAQ - in particular the "make menuconfig" and "make clean" steps.  If you haven't skipped those steps, then it sounds like something got corrupted - you may want to backup your printer.cfg and do a clean install of OctoPi to your sd card.

-Kevin

dzao, 2019-04-08:
Thank you everybody, it´s working now. I reinstal klipper and update it and everything working without problem. I´m running on two SKR 1.3 boards and eight drivers with 10 stepper motors. I´m going test quad gantry leveling with four indenpendent motors.

closed 2019-04-09 by dzao

KevinOConnor, 2019-04-11:
@dzao - can you post your final working printer config file?

Thanks.
-Kevin

elliotf, 2019-06-06:
@dzao your final working printer config file would be useful for others. Would you please attach it to a comment (not past the contents into a comment)?