Klipper3d/klipper #1511: Feature request  : M556 Axis Compensation

Haggenti, 2019-04-06:
Could it be possible to implement this, like reprapfirmware does (right angle correction of z axis for example) ?
Thanks again for all this very good work.

KevinOConnor, 2019-04-13:
I'm guessing this is similar to the existing #258.  It's fine if someone wishes to add support for it.

-Kevin

KevinOConnor, 2019-04-21:
I'm going to close this, as with #258 it does not look like anyone is currently working on implementing it.

-Kevin

closed 2019-04-21 by KevinOConnor

Geramy, 2019-05-24:
Kevin, I have a Flashforge Dreamer and I converted it over it looks like as it prints left it gets skewed or the item prints out longer on the left side then it does on the right side, would this requested feature fix this if it was implemented?