Klipper3d/klipper #1513: Feature request: MMU2 cloned mechanics

mwr666, 2019-04-07:
Hello,
I'm trying to add multi material feature to my printer. The mechanics is the same as prusa mmu2 but its connected to smoothieboard.
The problem is that i need the tool changer to work with the same extruder pinout for 5 extruders (extruder, extruder1, extruder2, etc..). Shared heater works fine but step, dir, enable pins cannot be assigned the same because config loads with system error: "pin P2.3 used multiple times in config".
I need the tool changer commands to run the change filament code for mmu2.
Next is slicer that uses different gcode for different extruders
Right now my only option is some sort of g-code post processor that will change all tool changes code to redirect everything to "extruder 0".
Is it possible to add some sort of option into firmware like "shared_heater" for shared extruder pinouts???

mwr666, 2019-04-08:
[klippy.log](https://github.com/KevinOConnor/klipper/files/3053768/klippy.log)

KevinOConnor, 2019-04-14:
It's certainly possible to implement that - but it would require Python code to be developed.  There were a few people working on various mmu2 code branches - maybe @SergeyKrutyko or @trevjonez has more info.

-Kevin

trevjonez, 2019-04-14:
On my fork I have an mmu2_homing branch. the main issue I have had so far is that using the tmc2130 stallguard to home the idler drum is VERY inconsistent. I am running super cheap motors so maybe that plays into it as well. If you can tune your drivers just right, things start to work as expected and it will home the idler, selector then step through all 5 filament paths and home the filament positions by using the finda as an endstop for a homing move.

As is, it doesn't use the new manual stepper logic as I had started working on it before that was introduced. 

What is left is to figure out how to best modify the toolhead/gcode to "virtualize" the tools. I am thinking the behavior of the `T` gcode should probably be extracted and made pluggable such that they can be reassigned to any other sort of macro rather than whatever the current impl is.

mwr666, 2019-04-14:
> On my fork I have an mmu2_homing branch. the main issue I have had so far is that using the tmc2130 stallguard to home the idler drum is VERY inconsistent. I am running super cheap motors so maybe [...]

> It's certainly possible to implement that - but it would require Python code to be developed. There were a few people working on various mmu2 code branches - maybe @SergeyKrutyko or @trevjonez has more info.
> 
> -Kevin

I've made progress with mmu2 port without any change in the code.
Added custom commands to config. Klipper is configured as a single extruder machine.
Custom gcode commands are based on manual stepper. Tool names in change tool comands needs to have other names than T1, T2, etc because when using those standard naming You get errors from firmware having it configured only for one tool. I'm using slic3r with post script that changes all T0 to CT0 (T1 to CT1, etc). Also added in slic3r settings  some tool change gcode using [previous_tool] placeholder for PT0 (PT1 etc) command so the idler can know the last used tool.

So in conclusion the only thing i need right now to make it work without slic3r post script is a possibility to use T0, T1, etc naming in custom commands without having safety warnings and aborting the print when tool changing commands used with single extruder motor in config.

Another helpful thing for mmu2 setup would be a possibility to disable only the extruder motor with M18 E.

KevinOConnor, 2019-05-25:
It looks like the conversation has concluded on this.  I'm going to mark this ticket as inactive.

-Kevin

closed 2019-05-25 by KevinOConnor