Klipper3d/klipper #1514: Filament Jam Sensor (type on/off signal)

WindoC, 2019-04-07:
Hi, I am trying to port my 3D printer from Repetier-FW to Klipper. All thing is good except the filament jam detection module seem not supported by Klipper. I would like to know if there is a plan to support this kind of sensor? 

If no plan to support that in master. For a workaround, is there any way to to get the filament usage or E step value by an external application (ex: some python script)?

Or there are any document or example to explain how to implement new module in Klipper.

link my Filament Jam Sensor:
https://www.thingiverse.com/thing:3067904

Thanks!

KevinOConnor, 2019-04-14:
There are a couple of filament sensor modules currently in the code - see klippy/extras/filament_switch_sensor.py and klippy/extras/tsl1401cl_filament_width_sensor.py .  I expect new code would need to be written for your particular sensor.

-Kevin

WindoC, 2019-04-15:
I know, after study the module filament_switch_sensor and some try and error. I complete the coding for the sensor last weekend.
I have submitted the pull requests #1544 for this simple design filament jam sensor.

KevinOConnor, 2019-08-05:
I'm going to close this issue as it looks like the original question was answered.  We can discuss PR #1544 in that PR.

-Kevin

closed 2019-08-05 by KevinOConnor