Klipper3d/klipper #1515: BLTOUCH 3.0 query_probe always open and doesn't trigger z_stop

dreamcryer, 2019-04-08:
I broke my BLTouch so I got a new one which is version 3.0. My old one worked fine on Ender 3 with stock board. But the new probe seems to be always in open state so homing Z doesn't work. (I added pin_up_touch_mode_reports_triggered: False)

During homing z, when the pin is triggered (pushed up), it will immediately deploy again and the motor keeps going down. Manual testing touch_mode shows that the deployed pin would stay back once triggered.

query_probe command returns open all the time no matter what the pin state is. So does query_endstops. I simply swapped the z-endtop connection to old limit switch without changing the config and the query_endstops would return correct states.

[klippy.log](https://gist.github.com/dreamcryer/263d4316d2311b4b652e4a70bf05ad2f)

jannau, 2019-04-08:
You probably need to switch the endstop logic of the BLTouch v3.0 to 5V away from the new open drain default. See my work in progress branch https://github.com/jannau/klipper/tree/bltouch_v30.

For reference the [BLTouch v3.0 manual](https://docs.wixstatic.com/ugd/f5a1c8_77037f55e5d542309d9fc178165c9f3f.pdf) and [related Marlin issue](https://github.com/MarlinFirmware/Marlin/issues/13345)

dreamcryer, 2019-04-09:
@jannau I tested your fix and it didn't solve the problem. Somehow the pin now stays back when it touched my finger during z homing. But 
[klippy.log](https://github.com/KevinOConnor/klipper/files/3057101/klippy.log)
the motor still kept going. I wonder if it's related to the probe status reading and pin_up_touch_mode_reports_triggered setting. Without the setting, the probe status is always triggered. And with that setting, probe status is always open.

dreamcryer, 2019-04-10:
I tried different ways to switch the probe to 5v mode in bltouch.py but wasn't able to get homing to work. The pin value never gets updated henece the probe not triggered. I added 1k ohm resistor between the z-min wire (white) and the 5v wire (red) as suggested in [this post](https://github.com/MarlinFirmware/Marlin/issues/13345#issuecomment-475924803) and it worked without the need to change printer.cfg.

According to the Marlin issue thread @jannau pointed out, the signal window shrank but it is still within Marlin's polling cycle (5ms). And Marlin's fix works with Ender 3. I think Klipper is polling the pin every 1ms if I didn't read the code wrong. So I am guessing somehow the probe was not switched to 5v mode correctly.

I know software but not enough about the electronics so hopefully someone could give a hand here.

KevinOConnor, 2019-04-14:
FWIW, I don't have any insight into the root cause of your issue.  I'd double check the wiring/config and follow the steps in docs/BLTouch.md .  You can put the probe in "touch mode" and verify it reports triggered on a query_probe when you manually press the probe.  (You can't detect a trigger this way in "pin down" mode.)  If the probe is sensing the trigger but not reporting it then I'd guess it's bad wiring/config or a broken probe.

-Kevin

Scy1, 2019-04-15:
I had the same issue with BLtouch 3.0 on my ender 3 and marlin. Ended up solving it following this advice on the link below and it worked.

https://www.reddit.com/r/ender3/comments/b59soj/psa_new_bltouch_v30_will_not_work_with_marlin_11x/ejf3q81?utm_source=share&utm_medium=web2x

Trying to switch to klipper now and it's the same problem, dunno how I would implement this in klipper. He says they fixed it by setting the probe in SW_MODE when deployed.

dreamcryer, 2019-04-15:
@KevinOConnor, yeah I tried touch_mode + query_probe before but klipper didn't report it triggered. I now added a resistor to work around it. I plan to switch to skr v1.3 soon. When I do, I will try again and report back. Hopefully it's not my probe.

@Scy1, see jannau's post above. He has a fix that he is working on. I am curious if it works for you. If it does, I know my wiring or probe is bad and we can also confirm that jannau's fix works.

Scy1, 2019-04-15:
Before the bltouch would hit the bed and keep moving down till the nozzle would depress the bed a bit and move back up and blink red. After I tried the fix the bltouch would hit the bed turn red, but the nozzle would keep going down and not stop till i turn of the printer.

dreamcryer, 2019-04-15:
>  After I tried the fix the bltouch would hit the bed turn solid red, but the nozzle would keep going down and not stop till i turn of the printer.

Exactly what's happening to me after I apply the fix. Looks like we do have some other things needed to be fixed in klipper.

KevinOConnor, 2019-04-21:
FYI, there appears to be a lot of similarity between this issue and issue #1483.

-Kevin

elcojacobs, 2019-04-25:
I have a BLTouch V3 from Creality direct, arrived last week. It also seems to be triggered at all times and doesn't work as a z-stop.

I think the issue of open drain being the default should be easy to fix by enabling the internal pull-up resistor of the AVR on the pin. That would not require putting the BLTouch in 5V mode.
That won't even affect non-open drain BLTouch with 5V output, because the pull-up is weak and easy to override by an external signal.

CircuitSetup, 2019-05-08:
I'm also having issues with the bltouch v3

Initially when homing, the Z axis would just sit there like the trigger was already initiated. After changing "pin_up_touch_mode_reports_triggered:" to false, the needle will now deploy, but when triggered, the red light will turn off, and the blue will come on very faintly. The Z axis will not stop though.

CircuitSetup, 2019-05-08:
Here is my log after running the bltouch test: 
[klippy.log](https://github.com/KevinOConnor/klipper/files/3155099/klippy.log)

vicentini, 2019-05-08:
I've also spent several hours trying to get my BlTouch to work. Here's what I ended up with.

My hardware
Ender 3 with PCB V1.1.4
Original BlTouch V3.0 (March 2019)

**1st Method** (same as @CircuitSetup)
- Use the original z stop pin (^PC4) as sensor pin
- Remove C7 capacitor
- Set pin_up_touch_mode_reports_triggered: False

**2nd Method**
Because we already use the ISP header for supply (GND, +5V) and control pin (MOSI-PB5), we can
- Use SCK pin with internal pull-up activated (^PB7) as sensor pin
(somehow MISO-PB6 pin was not working) 
- Set pin_up_touch_mode_reports_triggered: False

This way there is no need to remove any capacitors from the PCB, the original z stop can be left untouched and we can connect the BlTouch only to the ISP header. There's also no need for the "sensor_mode: logic_5v" modification as suggested by @jannau .

So far I only tested it with homing command G28, but it should also work for real bed leveling workflow, I suspect.

CircuitSetup, 2019-05-09:
Yep, mine is working fine for bed leveling. The only thing I noticed is when the pin retracts, the blue LED is on for a second, then goes out. It doesn't seem to effect anything though.

fietstasss, 2019-10-29:
> I've also spent several hours trying to get my BlTouch to work. Here's what I ended up with.
>
> My hardware
> Ender 3 with PCB V1.1.4
> Original BlTouch V3.0 (March 2019)
>
> **1st Method** (same as [...]

I don't fully understand how to wire for method #2, which has my preference. 
What do you mean with _"Because we already use the ISP header for supply (GND, +5V) and control pin (MOSI-PB5)"_   

I'm currently using the D11 row on my V2.1 CR-10S board, and was using the Z-HomeSwitch port for the black/white wires, but also faced the issue of having a 'always-open' reply from QUERY_PROBE. 
Therefore, I'd like to use your solution but can't find how to wire this.  Could you help me out with this? 

Thanks in advance!

sixtyfive, 2021-04-10:
For anyone coming here from Google: using the ISP header's MOSI and SCK pins for control and sensor does not work (anymore?) because Klipper (now?) knows what they are and will error out with `error: pin PXn is reserved for spi`. Remaining options are to find and remove C7 and keep using the original z endstop pin, or to find another one. The display header still has one available and there's "pin 27 adapters" you can buy, but they add quite a bit of height to the board. A better option might be the connectors labelled "CHECK" (left pin is PA2) and "OFF" (right pin is PC1).

vicentini, 2021-05-17:
@sixtyfive Just to make this clear and don't lead anyone on the wrong track.
The 2nd method (ISP header) I described above still works with the latest version of klipper. I still use the same hardware as stated above. Maybe you have a newer incompatible board, other version of BlTouch or just configured it the wrong way, but it is certainly still working.
If you claim that it's not working for you, please let the people know what setup you use.

sixtyfive, 2021-05-18:
If anything I have an older board. But is it also possible that it's down to Klipper version? SPI pins changing their pin numbers sounds unlikely.