Klipper3d/klipper #1516: Printers LCD no Beep Sound

novio8, 2019-04-08:
I have Ramps/ReprapDiscount LCD combo. Using printers Menu there is no Beep Sound when I press Click Button. Any ideas why, or how to enable Beep sound ?

[klippy.log](https://github.com/KevinOConnor/klipper/files/3055877/klippy.log)

Arksine, 2019-04-09:
Klipper's menu does not currently support audio feedback.

novio8, 2019-04-09:
OK.

shaiss, 2019-04-12:
@arksine. You can enable m300 in the config. That will then enable the beeper.  You'd have to customize the code to beep on menu clicks

Arksine, 2019-04-12:
Yes, it may possible to add a beep gcode for most every menu item.  However that probably isn't ideal, as I doubt most people want to re-implement the entire menu in printer.cfg.  Something like this would probably be better with a native solution, however I don't know if anyone is interested in doing so.

KevinOConnor, 2019-04-21:
I'm going to close this as I believe the original question has been answered (it's not currently implemented - it is possible to implement but that would require a developer interested in working on it).

-Kevin

closed 2019-04-21 by KevinOConnor