Klipper3d/klipper #1519: Ender 3 with new V1.1.4 mother board. It comes with TMC2208 drivers

JulioDavidMartinez, 2019-04-08:
Ender 3 with new V1.1.4 mother board. It comes with TMC2208 drivers
When I add the following to the config:
[tmc2208 stepper_x]
[tmc2208 stepper_y]
[tmc2208 stepper_z]

it asks me for a uart pin. But I have no idea what those would be for each driver.
If I put uart_pin: analog_5 the printer halts and will not reconnect.

JulioDavidMartinez, 2019-04-09:
Woops. Sorry about that. Attached please find the log.
[klippy.log](https://github.com/KevinOConnor/klipper/files/3056682/klippy.log)

jakep82, 2019-04-09:
From what I've been able to find out about this board, Creality didn't connect the UART pins from the drivers to the MCU. Likely because the MCU they're using has so few pins. This means the drivers are permanently in standalone mode, and as a result this board is not safe to use with Klipper.

JulioDavidMartinez, 2019-04-09:
OH NO!!!!!!!!!!!!! What a blow.... "Sherly Temple!" Darn! Darn! Darn!
I'm using it as is. Prints are successful. I never even imagined that the board would come with 2208s. If it wasn't for Teaching Tech's video today I would have left it alone. I have to say it's as loud as the A4988's the way it currently is. I would be curious to flash Marlin back on it and see if there is any difference in noise.
I'm truly bummed out that it's not safe with Klipper.
I LOVE KLIPPER!!!!!!!!!!!!!!!!!!!!
I'm going to hope against hope that Kevin in his infinite genius and wisdom will come up with some solution that will allow this. 
Creality is going to be using this board moving forward so if it's not supported that's going to be a lot of unhappy people!

FHeilmann, 2019-04-09:
You need a hardware connection to make that work so there's no magic that can be done by klipper itself.

I'd still say you can use the board with klipper, however you won't be able to make any adjustments on the drivers, so they will just end up being dumb drivers like the A4988s. Just remove the [tmc2208] blocks from your config and see how it goes.

ShohninDmitriy, 2019-04-09:
https://github.com/KevinOConnor/klipper/issues/1365

jakep82, 2019-04-09:
@FHeilmann It's not safe to run 2208 drivers in standalone with Klipper.

https://github.com/KevinOConnor/klipper/issues/196
https://github.com/KevinOConnor/klipper/blob/master/docs/FAQ.md#my-tmc-motor-driver-turns-off-in-the-middle-of-a-print

Hywelmartin, 2019-04-09:
@FHeilmann I think @JulioDavidMartinez  is referring to this ["problem" ](https://github.com/KevinOConnor/klipper/blob/master/docs/FAQ.md#my-tmc-motor-driver-turns-off-in-the-middle-of-a-print)

JulioDavidMartinez, 2019-04-09:
I have the printer running with no [tmc2208 stepper_x] or any of the other drivers currently. The printer prints. Tonight I will remove the heat sink on two of the steppers to see what's under the hood. I may have the A4988 chips in there. I saw on Teaching Techs's video:
https://www.youtube.com/watch?v=_wEZOAFZhB4
That the new board he purchased ( v1.1.4) which is the same version of board I have now comes with the tmc2208. I purchased mine about 3 weeks ago however, so it's possible I don't actually have the tmc2208. These companies from China are known to change things and not document anything, nor provide spec sheets or anything while keeping the same revision numbers on hardware. So who know. I will find out tonight and post my findings though.

Klipper Team Thanks A Million for the hard work you guys do! 
I do love your firmware and I am a patron on patreon to help you guys continue your great work.
Much appreciated.

JulioDavidMartinez, 2019-04-12:
I can confirm that my board does NOT have TMC chips. So sad.
Here are pics:
http://aenprodfileshare.com/retrieve.php?share=ec0azq1ym1o6_rb6kecuhwqlbdny6kytjvbwxbz_r3juvm71pthccrl3nhpk15ju5p78sckickvtqez0u6mxet7824456xew08igd8mdhjjkb_ubp1yf3k2pd8z7hbce

http://aenprodfileshare.com/retrieve.php?share=ec0azq1ym1o6_rb6kecuhwqlbdny6kytjvbwxbz_r3juvm71pthccrl3nhpk15ju5p78sckickvtqez0u6mxet7824456xew08igd8mdhj_kl_uvprr23k2pd8z7hbce

manu7irl, 2019-04-15:
did you buy this board separately from creality or did it get shipped to
you with your ender3?
if it was shipped eith your printer when did you buy it?

JulioDavidMartinez, 2019-04-15:
I got it aftermarket from Amazon. It was tricky because I saw other advertisements of a v1.1.4 board on Amazon but the pics all showed either the v.1.1.3 board or the v.1.1.2 board which is from an ender 2. Yikes. Michael from Teaching Tech on youtube got his directly from Creality. And he stated that it was labeled as "an upgrade board".

Hm..

manu7irl, 2019-04-15:
can you check another driver on your board it seems that you checked on the
extruder maybe the extruder stays with a A4988 because they give more
torque. then on the rest of them they are tmc2208... Can you confirm?
I ordered 3 new a week ago I received them with rainbow lcd cable and
upgraded 1.1.4 mcu board. they are silent has hell. but I cannot check
anymore if they have tmcs because they got shipped to my customers.

JulioDavidMartinez, 2019-04-15:
Sure I will check tonight when I get home from work. I'll be honest, I don't think my board has the TMC chips considering how noisy my new board is. I took off the dampers in hopes that they were not needed. I was wrong. Illl update tonight with pics of the chip. I'll do the Y axis chip and take a pic of it and post it.

This is the link I got my board from:
https://www.amazon.com/gp/product/B07NVH6BCD/ref=oh_aui_search_asin_title?ie=UTF8&psc=1

It's listed as v1.1.3, but what I got in the box has the printed text of v1.1.4.
I figured why the heck not. If things are better on the board, why not.
I was happy not knowing that there are boards out there with TMC chips. If I truly don't have them, then I'm just bummed.

manu7irl, 2019-04-16:
thanks, waiting to hear from you.

JulioDavidMartinez, 2019-04-16:
Sorry. Got home super late and was too tired. I will look tonight and post.

JulioDavidMartinez, 2019-04-17:
Sorry it took me so long. I'm working on a bit project at work and have been putting in late hours.
Confirm on Y axis. A4988 steppers. So I'm assuming all steppers are A4988. E and Y have A4988. Picture posted here:
http://aenprodfileshare.com/retrieve.php?share=ec0azq1ym1o6_rb6kecuhwqlbdny6kytjvbwxbz_r3juvm71pthccrl3nhpk15ju5p78sckickvtqez0u6mxet7824456xhi08igd5pqhj_klcxhpryf3k2pd8z7hbce

http://aenprodfileshare.com/retrieve.php?share=ec0azq1ym1o6_rb6kecuhwqlbdny6kytjvbwxbz_r3juvm71pthccrl3nhpk15ju5p78sckickvtqez0u6mxet7824456xhi08igd5pqhj_kbfubphh23k2pd8z7hbce

KevinOConnor, 2019-04-28:
I'm going to close this as it looks like the original issue was resolved (steppers not actually tmc2208).

-Kevin

closed 2019-05-02 by KevinOConnor

ximon, 2020-01-14:
Would it possible to use the tmc2208 ender 3 board with clipper if the UART pins are connected to the pi itself instead of the creality board (which doesn't have enough pins)?

jakep82, 2020-01-14:
> @KevinOConnor - Would it possible to use the tmc2208 ender 3 board with clipper if the UART pins are connected to the pi itself instead of the creality board (which doesn't have enough pins)?

This is not supported.

ximon, 2020-01-17:
For reference: I got uart working using PB7, PB5 and PA0 on the atmega128p4 on the 1.1.3 board. Lost the use of the sd card though.