Klipper3d/klipper #1522: Feature Request: Keep steppers enabled when PAUSE is issued

dushyantahuja, 2019-04-10:
Now with the pause functionality enabled in Klipper - can we have an option not to disable steppers if the printer is paused using PAUSE command (e.g. when filament runs out).

Currently, if the filament runs out and you do not replace within the stepper time-outs, it is very difficult to resume the print. Any help would be appreciated.

KevinOConnor, 2019-04-14:
It's possible to change the idle_timeout gcode to not disable motors on a timeout (see config/example-extras.cfg for the details).  With the recent template support for gcode macros, it may even be possible to do this programmatically within the gcode sections on a user-by-user basis.

-Kevin

dushyantahuja, 2019-04-15:
Thanks, Kevin - had thought of that, however, I would like to keep idle_timeout enabled for normal use - just want to make sure that the steppers are not disabled if the printer is paused. 
Will have to look at the gcode macros, if you think that might work.

KevinOConnor, 2019-05-25:
It looks like the conversation has concluded on this topic.  I'm going to mark this ticket as inactive.  If someone wishes to implement it, that's great.

-Kevin

closed 2019-05-25 by KevinOConnor

KevinOConnor, 2019-07-01:
FYI, the ability to configure the idle timeout was added in PR #1700 .

-Kevin

slavedas, 2019-11-20:
In case anyone else has this issue the printer.pause_resume.is_paused variable will allow selective enforcement of idle_timeout.  I have the following definition in my config which worked to keep my bed heated and motors locked in place until I got to my printer a few hours later.

[idle_timeout]
gcode:
          {% if printer.pause_resume.is_paused %}
                    M104 S0
          {% else %}
                    TURN_OFF_HEATERS
                    M84
          {% endif %}

andmiller, 2019-11-26:
> In case anyone else has this issue the printer.pause_resume.is_paused variable will allow selective enforcement of idle_timeout. I have the following definition in my config which worked to keep my [...]

@slavedas I get a parsing error if I use this snippet, I am on the latest Klipper version as of today, the bracketed portion is what is failing, any idea what I am missing?
 -Thanks

slavedas, 2019-11-26:
I've had that problem copying code from here as well it doesn't carry the tabs well.

You need to tab indent everything after gcode: and then add another tab indentation to the lines between the {%...%} statements

slavedas, 2019-11-26:
@andmiller it should look like this, but I had to use spaces and your tab distance may be longer or shorter so don't copy this code, but in your editor, hit tab once or twice depending on which indent level you see here.
`
[idle_timeout]
gcode:
        {% if printer.pause_resume.is_paused %}
                M104 S0
        {% else %}
                TURN_OFF_HEATERS
                M84
        {% endif %}
`

andmiller, 2019-11-26:
@slavedas Yeah, that seemed to work, should of realized the indenting issue. Is there any documentation of the variables we can use, that would be useful to many I would think. Thanks again

slavedas, 2019-11-26:
@andmiller https://github.com/KevinOConnor/klipper/blob/master/docs/Command_Templates.md