Klipper3d/klipper #1523: Inconsistencies in bed mesh leveling with BLTouch

DrMacabre, 2019-04-10:
Hello,

I can't get any consistent results when probing with my bltouch. I didn't noticed anything unusual when using it in marlin.

I did a test with 4 samples per calibration and they don't seem very consistent, especially at the first and last point

My BLTouch is a clone btw, i have dual Z axis upgrade with antibacklash to minimize gaps on the Z axis 

Thanks for any help.

probe at 60.000,35.000 is z=0.837500
probe at 60.000,35.000 is z=0.635000
probe at 60.000,35.000 is z=0.342500
probe at 60.000,35.000 is z=0.322500
probe at 180.000,35.000 is z=0.317500
probe at 180.000,35.000 is z=0.467500
probe at 180.000,35.000 is z=0.530000
probe at 180.000,35.000 is z=0.570000
probe at 300.000,35.000 is z=0.690000
probe at 300.000,35.000 is z=0.677500
probe at 300.000,35.000 is z=0.695000
probe at 300.000,35.000 is z=0.690000
probe at 300.000,160.000 is z=0.580000
probe at 300.000,160.000 is z=0.632500
probe at 300.000,160.000 is z=0.647500
probe at 300.000,160.000 is z=0.675000
probe at 180.000,160.000 is z=0.730000
probe at 180.000,160.000 is z=0.732500
probe at 180.000,160.000 is z=0.737500
probe at 180.000,160.000 is z=0.750000
probe at 60.000,160.000 is z=0.782500
probe at 60.000,160.000 is z=0.812500
probe at 60.000,160.000 is z=0.802500
probe at 60.000,160.000 is z=0.832500
probe at 60.000,285.000 is z=0.827500
probe at 60.000,285.000 is z=0.827500
probe at 60.000,285.000 is z=0.825000
probe at 60.000,285.000 is z=0.822500
probe at 180.000,285.000 is z=0.747500
probe at 180.000,285.000 is z=0.735000
probe at 180.000,285.000 is z=0.705000
probe at 180.000,285.000 is z=0.702500
probe at 300.000,285.000 is z=0.627500
probe at 300.000,285.000 is z=0.535000
probe at 300.000,285.000 is z=0.525000
probe at 300.000,285.000 is z=0.447500

[klippy.log.zip](https://github.com/KevinOConnor/klipper/files/3062634/klippy.log.zip)

Arksine, 2019-04-10:
Are you probing at the same general coordinates as you did with Marlin?

FWIW I have a bltouch clone and experienced inconsistent homing on a machine I don't even use Mesh Leveling with, so I ditched it.

DrMacabre, 2019-04-10:
Yes, same coordinates. I kinda feel the homing with the BLTouch is also off in klipper, i’v been doing countless bed leveling by hands using the assistant the past couple of days and it never seem to be good. I went back to marlin to make sure i wasn’t hallucinating and i got good results. I’ve been using it for about 4 months now and almost never had to level my bed.

KevinOConnor, 2019-04-21:
FYI, nothing in the log indicates a problem.  To the best of our knowledge, the Klipper code works as intended.  We have received a lot of reports of quality issues on clones.  Only suggestions I know of is to reduce probing speed and/or increase the pin move time.

-Kevin

rlesac, 2019-04-21:
> Hello,
> 
> I can't get any consistent results when probing with my bltouch. I didn't noticed anything unusual when using it in marlin.

Are you sure that it's working fine with Marlin? I had issues with clones, two of them being unusable.
Try the M48 test and report back.

Here's the tests:
>>> M48 X110 Y110 P10 V2
SENDING:M48 X110 Y110 P10 V2
M48 Z-Probe Repeatability Test

1 of 10: z: 0.295
2 of 10: z: 0.315
3 of 10: z: 0.338
4 of 10: z: 0.357
5 of 10: z: 0.373
6 of 10: z: 0.390
7 of 10: z: 0.417
8 of 10: z: 0.413
9 of 10: z: 0.450
10 of 10: z: 0.450

Finished!
Mean: 0.379750 Min: 0.295 Max: 0.450 Range: 0.155
Standard Deviation: 0.051167

Here's more tests:
Mean: 0.745250 Min: 0.678 Max: 0.845 Range: 0.167
Standard Deviation: 0.054732

Mean: 0.971750 Min: 0.910 Max: 1.032 Range: 0.123
Standard Deviation: 0.039116

Mean: 1.218750 Min: 1.147 Max: 1.282 Range: 0.135
Standard Deviation: 0.040670

Mean: 1.298250 Min: 1.150 Max: 1.402 Range: 0.252
Standard Deviation: 0.082682

Mean: 1.437250 Min: 1.357 Max: 1.530 Range: 0.173
Standard Deviation: 0.064764

Mean: 0.460500 Min: 0.395 Max: 0.518 Range: 0.123
Standard Deviation: 0.044494

Mean: 0.499750 Min: 0.440 Max: 0.527 Range: 0.087
Standard Deviation: 0.022512

Mean: 0.549000 Min: 0.500 Max: 0.600 Range: 0.100
Standard Deviation: 0.033000

Mean: 1.792000 Min: 1.665 Max: 1.890 Range: 0.225
Standard Deviation: 0.067812

Mean: 1.968750 Min: 1.905 Max: 2.025 Range: 0.120
Standard Deviation: 0.040066

KevinOConnor, 2019-05-16:
I'm going to close this as it looks like the discussion has ended.  To the best of our knowledge, the bltouch code in Klipper does what is intended.

-Kevin

closed 2019-05-16 by KevinOConnor