Klipper3d/klipper #1525: Question, not an issue.

JulioDavidMartinez, 2019-04-11:
I've installed an AC heated bed that comes with the NTC 10k 3950 thermistor. How do I set this up correctly in firmware?
I've attached my klippy log for the sake of the bot.
Thanks for any help.
[klippy.log](https://github.com/KevinOConnor/klipper/files/3066646/klippy.log)

gdachs, 2019-04-12:
[heater_bed]
...
sensor_type: NTC 100K beta 3950

JulioDavidMartinez, 2019-04-12:
Thank you so much!!!

gdachs, 2019-04-12:
Wait a moment. I see only now that your heater uses a 10K Thermistor. That is very uncommon. My settings will not work with it. I have an AC heater too, but it uses a 100K thermistor.

jakep82, 2019-04-12:
The Amazon link says 100k.

JulioDavidMartinez, 2019-04-13:
The Amazon Link does say 100k. I made a typo. Sorry. However when I look online for that thermistor, oddly there are two versions. 100k and 10k. I'll assume that the Amazon link is the correct one.

Again thank you guys so much for all you do!!!!

KevinOConnor, 2019-04-21:
I'm going to close this as it looks like the original issue has been resolved.

-Kevin

closed 2019-04-21 by KevinOConnor