Klipper3d/klipper #1526: Probe pause - is this normal or something I can adjust?

thucar, 2019-04-11:
Using a BLTouch and there's one thing that irks me. Whenever the printer is probing, it pauses between samples for just a second or two.

So a Z-Home behavior would look like this:

1. Move to the designated spot, lower z until probe trips
2. Pause for a second
3. Raise Z a few mm
4. Pause for a second
5. Lower Z at lower speed until triggered

While for z-homing this is not too much of an issue, the pauses do add up when doing bed mesh leveling.

I'd love to achieve something like this: https://www.youtube.com/watch?v=qJJasdv60yA

KevinOConnor, 2019-04-14:
On Klipper, it's possible to customize the bltouch pin_move_time (see config/example-extras.cfg for the details).  That's the bulk of the probing delay.  Aside from that, due to the micro-controller / host split, there are some smaller probing delays in Klipper that are difficult to remove.

-Kevin

thucar, 2019-04-19:
Ok, that is what I suspected. Thanks.

closed 2019-04-19 by thucar