Klipper3d/klipper #1528: Move Exceeds Maximum Extrusion

floppyslotter, 2019-04-12:
It seems like my printer has been hit with the dreaded Maximum Extrusion error.  I was printing just fine yesterday but today after work it seemed to do nothing but throw this error.  My print completes the first layer and then once it starts the second layer it decides to trigger this maximum extrusion error.  I did look through some previous threads pertaining to this topic and experimented with some of the solutions outlined in those threads.  Adjusting or even completely turning off pressure advance has had no effect, no change with print speed, have tried adding in a dummy G1 command in my starting scripts with no change, no change when selecting or deselecting "sticky parameters" in S3D.  I am using S3D 4.1.1 as my slicer btw.  I have made no changes or updates to slicer software, octoprint, klipper config/firmware, etc within the last few months.  I think the next step for me to try is switching slicers and seeing what happens, I've been seeing some people state that S3D is generating weird G-code commands that may be resulting in this error.  Log file is attached below, I rarely shut off my rpi so the log may be a little long and have some other misc. errors from me farting around but the issue is the last thing saved in the log.  Maybe I should give my rpi a good old fashioned turn it off and on again?

Any help or input to help me resolve this would be greatly appreciated!

[klippy.log.zip](https://github.com/KevinOConnor/klipper/files/3071245/klippy.log.zip)

alfsoft, 2019-04-12:
Hi @floppyslotter,

It did not look like there was a Klipper log file attached to this tic... Oh, there it is! :laughing:

Yes, that's S3D's issue with G-Code weirdness. I tried to disable "Firmware supports "sticky" parameters" option in G-Code tab as @mzakelj stated [here](https://github.com/KevinOConnor/klipper/issues/948#issuecomment-444420126) and that worked for some time. Although I didn't like that my printer stuttered (paused for a second here and there midprint) with this setting off.
So I switched firmware stickiness on again and started to use my own recipe described [here](https://github.com/KevinOConnor/klipper/issues/461#issuecomment-418842623). I know that it's a really wrong way of fixing this issue but nothing bad happened since September 2018.

floppyslotter, 2019-04-13:
I increased my max extrusion value and then started to get yet a different error now; MCU shutdown: rescheduled time in the past.  So I incrementally lowered my max extrusion value back down until I am now able to successfully print GCode generated by S3D again.  

I see where you modified the kinematics/extruder python script but I personally feel too inexperienced trying to change mine just in case I accidentally bork something even further.

floppyslotter, 2019-04-21:
So to make a long story short, unplugging the LCD screen on my Ender 3 resolved my issue...for now.

I stopped using S3D for a while and switched over to Cura and Slic3rPE, both would generate gCode that would print on my Ender 3 without issue.  But occasionally I would get MCU Shutdown errors instead of maximum extrusion exceeded errors.  I googled and found that it may be related to either my USB cable or the power supply for my rpi.  I then hopped over to the Klipper discord and it was suggested to me there that I first unplug the LCD screen on my Ender 3 to cut down on current draw from the 5v supply of my USB cable from the rpi to my printer.  I did this and performed a few more prints without receiving another MCU shutdown error.  So for giggles I sliced a model in s3d and fed it to my printer, and it is currently printing out that model with no issues so far where before, the maximum extrusion error would always show up as the second layer began to print, every time, without fail.  I will try to continue updating if I run into another issue.

KevinOConnor, 2019-04-21:
It appears you are running Klipper code from about six months ago.  There's been a number of bug fixes in that time.  If you're having an issue I recommend you upgrade the software (see the FAQ for details).  If you continue to experience issues then attach the log file from that run.

-Kevin

closed 2019-05-16 by KevinOConnor