Klipper3d/klipper #1529: Cannot Tune from LCD while printing from vSD

Nitrock28, 2019-04-12:
Hi,

I started to use the virtual SD feature of Klipper and remarked that the "tune" menu does not work for me. When I change the Z height in the first layer, I don't see the currently displayed offset change. I also tried to change the flow rate with no more success. The displayed value remains unchanged during the print, and the printer behavior (quality of the first layer) shows that it is not a graphical bug, the values are not modified.

When restarting the print from octoprint without the SD, everything works fine and I can then tune the first layer. 

In the Log, the SD print in question is between lines 3006 and 3636, and the print after that is from octoprint.

P.S: I know didn't activate the respond config section, my gcode tries to send minutes remaining to octoprint periodically, I think that can be ignored.

[klippy.log.zip](https://github.com/KevinOConnor/klipper/files/3071986/klippy.log.zip)

KevinOConnor, 2019-04-21:
There have been reports that the virtual sd card can prevent some menu commands from working in a timely manner.  I believe @arksine reported this as well, but I don't have the github id handy.  So, I believe this is a defect in Klipper.

-Kevin

Nitrock28, 2019-04-23:
I think I found the other thread you are talking about : #1236. 
I will stop using the virtual SD card for now, Thanks for the feedback !

KevinOConnor, 2019-06-09:
This should now be fixed (commit 962f7b98).

-Kevin

closed 2019-06-18 by KevinOConnor