Klipper3d/klipper #1531: G-Code move transform already specified

james-golden, 2019-04-12:
I have just did an update to the printer, did the git pull and had to re-flash the controller.  It worked once and then I tried to get the bed mesh working and I keep receiving the error "G-Code move transform already specified" and it disconnects.  Log attached
[klipper.log](https://github.com/KevinOConnor/klipper/files/3073459/klipper.log)

james-golden, 2019-04-12:
Ok, I think that I found it.  There is a bed tilt specified in this part of my config.txt file.  This was generated when I did the full delta calibration with the printed part.  Should i comment it out or is there another recourse
#*# <---------------------- SAVE_CONFIG ---------------------->
#*# DO NOT EDIT THIS BLOCK OR BELOW. The contents are auto-generated.
#*#
#*# [printer]
#*# delta_radius = 130.711717
#*#
#*# [stepper_a]
#*# arm_length = 261.229648
#*# angle = 89.098796
#*# position_endstop = 290.085868
#*#
#*# [stepper_b]
#*# arm_length = 264.000028
#*# angle = 209.558616
#*# position_endstop = 288.345379
#*#
#*# [stepper_c]
#*# arm_length = 260.946501
#*# angle = 330.000000
#*# position_endstop = 290.110099
#*#
#*# [bed_tilt]
#*# x_adjust = 0.000398
#*# y_adjust = -0.000751
#*# z_adjust = 0.105784
#*#
#*# [extruder]
#*#
#*# [delta_calibrate]
#*# height0 = 0.2
#*# height0_pos = 46310.000,46166.000,46331.000
#*# height1 = 0.2
#*# height1_pos = 41338.000,53272.000,53579.000
#*# height2 = 0.2
#*# height2_pos = 44962.000,44705.000,57969.000
#*# height3 = 0.2
#*# height3_pos = 52491.000,41487.000,52419.000
#*# height4 = 0.2
#*# height4_pos = 54430.000,44728.000,44845.000
#*# height5 = 0.2
#*# height5_pos = 51332.000,51260.000,41991.000
#*# height6 = 0.2
#*# height6_pos = 44796.000,55867.000,44886.000
#*# distance0 = 64.29
#*# distance0_pos1 = 46661.492,45780.899,46451.916
#*# distance0_pos2 = 51790.433,41801.254,51382.457
#*# distance1 = 64.71
#*# distance1_pos1 = 46870.301,45987.954,46045.113
#*# distance1_pos2 = 55796.169,44794.321,44777.077
#*# distance2 = 64.66
#*# distance2_pos1 = 46657.311,46394.777,45843.348
#*# distance2_pos2 = 51715.271,51415.347,41863.181
#*# distance3 = 63.93
#*# distance3_pos1 = 46239.183,46594.465,46045.113
#*# distance3_pos2 = 44908.734,55190.432,44777.077
#*# distance4 = 64.39
#*# distance4_pos1 = 46033.957,46384.043,46451.916
#*# distance4_pos2 = 41956.717,51224.301,51382.457
#*# distance5 = 64.05
#*# distance5_pos1 = 46243.317,45977.335,46656.990
#*# distance5_pos2 = 44970.377,44635.652,55253.607
#*# distance6 = 64.14
#*# distance6_pos1 = 51426.854,42085.124,50345.098
#*# distance6_pos2 = 55377.126,45091.405,44486.376
#*# distance7 = 64.04
#*# distance7_pos1 = 54563.026,45089.828,44489.947
#*# distance7_pos2 = 51112.197,51529.674,41956.011
#*# distance8 = 64.75
#*# distance8_pos1 = 50642.826,51056.902,42142.293
#*# distance8_pos2 = 44612.864,54787.238,45073.587
#*# distance9 = 64.29
#*# distance9_pos1 = 44618.498,54010.757,45077.212
#*# distance9_pos2 = 42053.432,50644.263,51502.677
#*# distance10 = 64.11
#*# distance10_pos1 = 42243.827,50195.935,51037.302
#*# distance10_pos2 = 45274.796,44350.639,54857.312
#*# distance11 = 64.53
#*# distance11_pos1 = 45276.502,44359.272,54080.596
#*# distance11_pos2 = 51913.019,41898.856,50800.696

james-golden, 2019-04-12:
ok, did a reboot of octoprint with the bed mesh section removed and it is back up and working.  So is the save file part of the printer config not compatable with using bed mesh?  I had commented out the delta calibrate portion before trying the bed mesh on my delta.

james-golden, 2019-04-12:
Hit the wrong button

Arksine, 2019-04-12:
Bed Tilt and Bed Mesh cannot be used together.  If you have bed_tilt saved in the persistent section you need to remove it to use bed_mesh.

james-golden, 2019-04-12:
Thank you, That is what I thought I would need.  Will that mess with the delta calibration that I have saved there? Or should I remove the persistent section?

Arksine, 2019-04-12:
If you just remove the [bed_tilt] section it shouldn't affect delta calibrate.  If you remove the whole thing then it you will also remove your delta calibration.

Its probably wise to make a backup copy of your current printer.cfg before making changes (as its typically not advised to edit the persistent section).  However if you just remove bed_tilt you should be okay.

james-golden, 2019-04-13:
That took care of it. Thank you for all the help

closed 2019-04-13 by james-golden