Klipper3d/klipper #1533: BLTouch does not work

JulianOhm, 2019-04-12:
I use an Ender 3 with stock Hardware so far. I bought the creality upgrade kit with original BLTouch.

Wiring seems to be good. BLTouch is doing his self test and then the light ist on and pin up. I can set pin_down and pin_up. Works as aspected.

But on G28 while trying to stop z axis with my fingernail or a piece of folded paper, the pin goes up, then down again. i tried to stop 4 times but nothing happens.

There are some informations that i should change the black and white cable. i tried that, but nothing happens. some other errors. pin does not work as aspected. so looks like the wiring should be okay.

is there a command to test the value of the pin? pin state or something like that?

[klippy.log](https://github.com/KevinOConnor/klipper/files/3074805/klippy.log)

jannau, 2019-04-13:
Is the BLTouch be any chance a BLTouch v3.0? If yes see #1515.

JulianOhm, 2019-04-14:
No i checked this. It's a V2.2

KevinOConnor, 2019-04-14:
You are running code from a few months ago - there were several important fixes in that time - I recommend upgrading to the latest code (see the FAQ for details) and retrying.

-Kevin

JulianOhm, 2019-04-14:
[klippy.log](https://github.com/KevinOConnor/klipper/files/3077706/klippy.log)

okay i upgraded the host and firmware but i get a new error: no jinja2....

JulianOhm, 2019-04-14:
okay.. my mistake.. forgot to start the install script

JulianOhm, 2019-04-14:
[klippy.log](https://github.com/KevinOConnor/klipper/files/3077765/klippy.log)

okay here the new klippy.log
still the same error. i tried to hit the pin 5-6 times with my piece of cardboard. But still the printer does not stop

JulianOhm, 2019-04-20:
no idea?

KevinOConnor, 2019-04-21:
It does not appear anyone has any advice.  Nothing in the log indicates an error.  Typically a situation like this would be indicative of the sensor_pin being incorrect.  However, there have been reports of recent genuine bltouch devices being sensitive to capacitors - see issue #1483.

-Kevin

JulianOhm, 2019-04-21:
It works now. I changed the cables back to the origin and not like it's in the manual. If someone uses the Creality original BLTouch set: Do not use the wiring diagram in the manual! The cable comes with right pin setting.

Thanks all for help!

closed 2019-04-21 by JulianOhm