Klipper3d/klipper #1534: MCU/Board list

timmit99, 2019-04-13:
Hi Kevin,
This isn't so much of an Issue as it is a request. I've been getting a lot of questions regarding what boards Klipper works on. Is there a list of boards that are known to work fine/work with some intricacies? If there isnt i would be willing to start putting one together as i tihnk it would be nice to have a chart to point people to to see what boards work/to get.
Something like the image below from the Marlin 2.0 32-bit thread:
[image]

Thanks for the amazing firmware and great work!

KevinOConnor, 2019-04-14:
The best we have at this point is the Features page: https://github.com/KevinOConnor/klipper/blob/master/docs/Features.md#step-benchmarks .  The config directory is also a useful reference: https://github.com/KevinOConnor/klipper/tree/master/config .

-Kevin

timmit99, 2019-04-15:
I was thinking something like this:

Hosts
-----

|       Board       |       MCU       |
|-------------------|-----------------|
| Raspberry Pi 3    |    BCM2837B0    |
| BBB with Replicape|     AM3358      |

MCUs
-----

|      Board      |       MCU       |
|-----------------|-----------------|
| Arduino Mega    |    ATMega2560   |
| RAMBo 1.4       |    ATMega2560   |
| RUMBA           |    ATMega2560   |
| Duet Wifi       |     SAM4E8E     |
| Sanguinololu    |    ATmega644P   |
| SmoothieBoard   |     LPC1769     |

smark-, 2019-04-19:
@timmit99 I think it's a great idea! I'm up to contribute if you set up the wiki page

KevinOConnor, 2019-04-20:
The closest thing in Klipper is the file listing in the config directory ( https://github.com/KevinOConnor/klipper/tree/master/config ).  It's fine if someone wishes to add a document to the docs/ directory and agree to maintain it.  One challenge is that Klipper supports a _lot_ of boards.  The same Klipper code (95% +) is used on every board, so generally speaking, if Klipper can run at all it will run fully.

-Kevin

sniperlucian, 2019-05-06:
I second @KevinOConnor  here. But it is actually several levels:

Firmware at the board:

1)  supported with MCU (ATmegaXXX, LPCXXX, STM32xxx)

Configs at Host/Master computer:

2)  supported config file for EACH board having the Pin's defined
3) supported config file for special functions of this board (SPI for stepper etc.)
4) supported config file for a special printer (Anet, Prusa etc.)
   
That gets fast very complex, as features are also very mixable (e.g. like 6 different drivers etc.). Level 4 certainly the easiest for noobs, as the others needs a lot of infos put together and the documentation of klippy still very abstract at some points. 

Maybe an open Google Xsheet, where people can integrate they successful implementations? Best with an link to the config file.

KevinOConnor, 2019-05-25:
It looks like the conversation has concluded on this topic.  If someone wishes to take this up then please add the document to the docs/ directory and create a github pull request for it.

-Kevin

closed 2019-05-25 by KevinOConnor