Klipper3d/klipper #1536: Random Microstep change

SimoneBnc, 2019-04-13:
Hello

I am experiencing random microstep change I suspect as the printer starts moving twice as fast and twice the distance (al least I noticed it for X and Y) and of course it is hitting everything as it is moving. Here is the log of the latest event 
[klippy.log](https://github.com/KevinOConnor/klipper/files/3075909/klippy.log)

KevinOConnor, 2019-04-14:
FYI, if motor power is removed from a trinamic driver (tmc2208, tmc2130, etc.) then they lose their internal state.  So, if that power is removed and then applied again, it will be necessary to run a RESTART command (or perhaps a series of INIT_TMC commands).  If you don't think this is the root cause, you could try issuing DUMP_TMC commands after an anomaly to see what the driver state is.

-Kevin

SimoneBnc, 2019-04-14:
I had the same issue the following print after I used the emergency stop and then restarted the firmware. I haven't found a common situation that triggers the problem. I do power down the main power supply between prints but I normally don't have any problems as the controller board stays on as it is powered via USB. I'll try the dump_tmc but I still don't understand how they could get powered down

KevinOConnor, 2019-04-14:
The tmc stepper driver aren't powered from the micro-controller power - they are powered from the main power supply.  If you turn off the main power supply you will need to manually issue a RESTART (or INIT_TMC as above) each time you turn the power back on.

-Kevin

SimoneBnc, 2019-04-14:
But then it would be a recurring issue and I would have noticed it. I'll try a few more tests. Dump_tmc should not work if the drivers are powered down, right?

SimoneBnc, 2019-04-15:
Anyway, I can run the dump_tmc even with main power supply off (I am using the Fysetc F6 board)

KevinOConnor, 2019-04-16:
@SimoneBnc - if motor power is removed, you'll likely get random results.  (Sometimes the chip will reset, sometimes registers will stay the same, sometimes registers will have some corruption.)

-Kevin

closed 2019-05-02 by KevinOConnor