Klipper3d/klipper #1537: Printer is not ready, Ender 3 Pro

Helme87, 2019-04-13:
Hello everyone
i Have installed Klipper on my ender 3 Pro, with a Raspberry b+.
after first issues. i could resolve :)

is there a new one.... 

Printer is not ready
The klippy host software is attempting to connect. Please
retry in a few moments.
Klipper state: Not ready

maybe someone has an idea what to do?

thanks for the help

alfsoft, 2019-04-13:
Check that your printer do really have this name in [mcu] section of printer.cfg:

serial = /dev/serial/by-id/usb-FTDI_FT232R_USB_UART_AK06N3EO-if00-port

This is the root of your problem:
`Unable to open port: [Errno 2] could not open port /dev/serial/by-id/usb-FTDI_FT232R_USB_UART_AK06N3EO-if00-port: [Errno 2] No such file or directory: '/dev/serial/by-id/usb-FTDI_FT232R_USB_UART_AK06N3EO-if00-port'`

To check real device name just log into your RPi via Putty on Windows or by using SSH on Linux and execute the following command:
`ls /dev/serial/by-id/`

This will list you devices present in this /dev/serial/by-id/ directory. Your command result should look like this:
```
pi@octopi:~ $ ls /dev/serial/by-id/
usb-FTDI_FT232R_USB_UART_AH06PMQV-if00-port0
```
In my case **usb-FTDI_FT232R_USB_UART_AH06PMQV-if00-port0** is the name of my board.

Helme87, 2019-04-13:
I tried that, and had to see that ther whas an writing error...
i checked the serial port again with ls /dev/serial/by-id/

but probelme is the same...
[klippy.log](https://github.com/KevinOConnor/klipper/files/3076167/klippy.log)

closed 2019-04-13 by Helme87