Klipper3d/klipper #1542: 3D Touch/BLTouch on Ender 3 - 3 issues and 1 question

JoaoFelipe, 2019-04-13:
I'm having some issues to install the 3d touch that I just received. 

Logs: 
[klippy.logs.zip](https://github.com/KevinOConnor/klipper/files/3076741/klippy.logs.zip)

## Issue 1 - When I home, the nozzle stays ~2mm above the bed, no matter the configuration I use.

I follow the guides to configure it, it seems to be working, but I don't seen any difference when I home.

See klippy.log in the attached file for this issue

Steps I followed (the issues start at the step 8 and are marked in bold):
1- Configured ~/printer.cfg according to https://github.com/KevinOConnor/klipper/blob/master/docs/BLTouch.md
```
[stepper_z]
endstop_pin: probe:z_virtual_endstop

[bltouch]
sensor_pin: ^PC4
control_pin: PA4
x_offset: 50.5
y_offset: 16

[homing_override]
set_position_z: 0..0
gcode:
    G90
    G1 Z10
    G28 X Y
    G1 X115 Y115 F6000
    G28 Z
```

2- Deleted the previous klippy.log and restarted klipper
```
$ rm /tmp/klippy.log
$ sudo service klipper restart
```

3- Connected in the Octoprint

4- Sent `BLTOUCH_DEBUG COMMAND=pin_down` and the pin went down

5- Sent `BLTOUCH_DEBUG COMMAND=pin_up` and the pin went up

6- Started following the instruction on https://github.com/KevinOConnor/klipper/blob/master/docs/Probe_Calibrate.md

7- Sent `G28` (Home). The nozzle went to the middle of the bed (position 115, 115) and ~2mm above the bed

8- Sent `PROBE`. **The pin tried to go down, but there was no space. Received the following errors**:
```
Recv: // Failed to verify BLTouch probe is raised; retrying.
Recv: // Failed to verify BLTouch probe is raised; retrying.
```

9- Used the control panel to move the nozzle up (`G1 Z10 F200` twice)

10- Sent `PROBE` again. Now, it did move down:
```
Recv: // probe at 115.000,115.000 is z=1.267500
```

11- Marked the position it touched and sent `GET_POSITION`:
```
Recv: // mcu: stepper_x:-5649 stepper_y:-237 stepper_z:-11270
Recv: // stepper: stepper_x:115.000000 stepper_y:115.000000 stepper_z:1.267500
Recv: // kinematic: X:115.000000 Y:115.000000 Z:1.267500
Recv: // toolhead: X:115.000000 Y:115.000000 Z:1.267500 E:0.000000
Recv: // gcode: X:115.000000 Y:115.000000 Z:1.267500 E:0.000000
Recv: // gcode base: X:0.000000 Y:0.000000 Z:0.000000 E:0.000000
Recv: // gcode homing: X:0.000000 Y:0.000000 Z:0.000000
Recv: ok
```

12- Sent `G1 X64.5 Y99 Z15. The nozzle pointed to the probe position. The configured x_offset and y_offset of my printer.cfg were right (I measured it before). Nonetheless, I issued a `RESTART`

13- Sent `PROBE_CALIBRATE`. Received an expected error
```
Recv: !! Error during homing move: Must home axis first: 0.000 0.000 -2.000 [0.000]
```

14- Sent `G28`. The nozzle went to he middle of the bed again. **It did not adjust according to x_offset and y_offset**

15- Sent `PROBE_CALIBRATE`. **Once again, the pin tried to go down, but it couldn't and I received an error:**
```
Recv: // Failed to verify BLTouch probe is raised; retrying.
Recv: // Failed to verify BLTouch probe is raised; retrying.
```

16- Sent `G1 Z20 F200`

17- Sent `PROBE_CALIBRATE`. Now the probe went to the middle of the bed
```
Recv: // probe at 115.000,115.000 is z=1.242500
Recv: // Starting manual Z probe. Use TESTZ to adjust position.
Recv: // Finish with ACCEPT or ABORT command.
Recv: // Z position: ?????? --> 6.242 <-- ??????
Recv: ok
```

18- Sent a bunch of TESTZ commands, performing the paper test:
```
Send: TESTZ Z=-5.5
Recv: // Z position: ?????? --> 0.743 <-- 6.242
Recv: ok
[...]
Send: TESTZ Z=-0.5
Recv: // Z position: ?????? --> 0.243 <-- 0.743
Recv: ok
[...]
Send: TESTZ Z=-0.2
Recv: // Z position: ?????? --> 0.043 <-- 0.243
Recv: ok
```

19- Sent `ACCEPT`, `SAVE_CONFIG`, and `RESTART`:
```
Send: ACCEPT
Recv: // bltouch: z_offset: 1.200
Recv: // The SAVE_CONFIG command will update the printer config file
Recv: // with the above and restart the printer.
Recv: ok
[...]
Send: SAVE_CONFIG
Recv: ok
Recv: // Klipper state: Disconnect
[...]
Recv: // Klipper state: Ready
[...]
Send: RESTART
Recv: ok
Recv: // Klipper state: Disconnect
[...]
Recv: // Klipper state: Ready
```

20- Tried to check the configuration by homing (`G28`). **Once again, the nozzle went to the middle of the bed (115, 115), staying ~2mm above it. The z_offset configuration does not seem to have any effect**

I tried some variations of these steps, with no success

## Issue 2 - Alert state after turning off the printer with the proble close to the bed

Steps:

1- Turned off the printer

2- Turned on the printer. The probe goes down and up twice, touching the bed and the led stays red 

3- Sent `FIRMWARE_RESTART`. **Received BLTouch errors** and the led started to blink:
```
Send: FIRMWARE_RESTART
Recv: ok
Recv: // Klipper state: Disconnect
[...]
Recv: // Failed to verify BLTouch probe is raised; retrying.
[...]
Recv: // Failed to verify BLTouch probe is raised; retrying.
Recv: // Klipper state: Ready
```

4- To exit this state, I have to send `G28`

## Issue 3 - Sometimes the probe does not go down.

I'm not sure this is an issue with Klipper or with my 3dtouch, but sometimes, when I try to home, the probe pin does not go down and the home command does not complete.

See klippy2.log in the attached file for this issue.

However, when I send `BLTOUCH_DEBUG COMMAND=pin_down` it works. 

I don't know how to solve this one reliably. I tried sendind `BLTOUCH_DEBUG COMMAND=reset`, but it didn't work at the next `G28`. However, when I tried to send `G28` again, without any changes it did work.

## Question - Can I use 3d touch bed mesh leveling, but keep the original z endstop for precision?

Due to all these issues, the 3d touch does not appear to be very reliable to me. Is it possible to use the original z endstop alongside the 3d touch? What should I configure? Would the following configuration work:
```
[stepper_z]
endstop_pin: ^PC4

[bltouch]
#sensor_pin: ^PC4
...
```

JulianOhm, 2019-04-14:
Looks like your homing override Gas an error. Set Position z should be 0.0 Not 0..0

KevinOConnor, 2019-04-14:
Your homing_override (the one in the log files) indicates a request was made to move the Z up 10mm at the start of the print.  Make sure your z step_distance parameter is correct, and in general make sure the Z is working correctly (see https://github.com/KevinOConnor/klipper/blob/master/docs/Config_checks.md ).  Make sure the Z correctly moves 10mm up at the start of G28 commands before going on the diagnose the 3dtouch.

Yes, you can use an endstop for homing and the probe for mesh.  Set the stepper_z endstop_pin to your endstop pin to do this.  You will need to wire the endstop and bltouch to different pins on the micro-controller.

-Kevin

JoaoFelipe, 2019-04-22:
The set_position_z was right in the config files. I just failed to copy it here.

I'm not sure about the z step_distance, I took it from the ender 3 files. I changed it to `.0125`, as it is the same step_distance of the other steppers, but I didn't notice any difference.

Everything in the config checks works.

The Z correctly moves 10mm up at the start of G28. It does override the homing and works "as expected". I think the main problem is that it does not take the configured probe z_offset into account. Or it does take it into account, but I'm configuring it wrongly because I receive errors when I try to send `PROBE` and `PROBE_CALIBRATE`

I just tried to install a BFPTouch (https://www.thingiverse.com/thing:2890290/), and I still have most of the problems of issue 1:

New config:
```
[stepper_z]
step_distance = .0125
endstop_pin: probe:z_virtual_endstop
#position_endstop = 0.0
#endstop_pin = ^PC4
position_min = -2

[homing_override]
gcode:
    G90 ; Use absolute position mode
    G1 Z10 ; Move up 10mm
    G28 X Y
    G1 X166 Y120 F6000 ; Change the X and Y coordinates to the center of your print #bed
    G28 Z
set_position_z: 0.0

[servo BFPTouch]
pin: PA4
initial_angle: 55
enable: True
maximum_servo_angle = 90
minimum_pulse_width = 0.001
maximum_pulse_width = 0.002

[probe]
pin: ^PC4
speed: 5.0
x_offset: -50
y_offset: 16
#z_offset: 1.0
activate_gcode:
    SET_SERVO SERVO=BFPTouch ENABLE=1
    SET_SERVO SERVO=BFPTouch ANGLE=90
    G4 P300
deactivate_gcode:
    SET_SERVO SERVO=BFPTouch ANGLE=55
    SET_SERVO SERVO=BFPTouch ENABLE=0

#*# <---------------------- SAVE_CONFIG ---------------------->
#*# DO NOT EDIT THIS BLOCK OR BELOW. The contents are auto-generated.
#*#
#*# [probe]
#*# z_offset = 0.800
```

Problems

8- Sent `PROBE` after `G28` . **The pin tried to go down, but there was no space. Received the following error**:
```
!! Probe triggered prior to movement
```
Followed steps 9-12 to configure x_offset and y_offset.

14- Sent `G28`. The nozzle went to he middle of the bed again. Different from the previous results the nozzle did consider the x_offset and y_offset this time

15- Sent `PROBE_CALIBRATE`. **Once again, the pin tried to go down, but it couldn't and I received an error**:
```
!! Probe triggered prior to movement
```

16-19- Sent `G1 Z20 F200`, `PROBE_CALIBRATE`, `TESTZ`, `ACCEPT`, `SAVE_CONFIG`, `RESTART`. Everything worked as expected.

20- Tried to check the configuration by homing (G28). The probe went to the middle of the bed, but the nozzle still stays ~1mm above

I don't have issues 2 and 3 with the BFPTouch, so I think I have a deffective 3dtouch

jakep82, 2019-04-22:
You need to include a Z movement in your activate g-code.

```
activate_gcode:
    SET_SERVO SERVO=BFPTouch ENABLE=1
    SET_SERVO SERVO=BFPTouch ANGLE=90
    G1 Z5
    G4 P300
```

JoaoFelipe, 2019-04-22:
This command made the head go right into the bed without stopping and without even releasing the pin. I tried to move the commando to after the SET_SERVE commands: It released the pin, but the head also went into the bed. I guess it does not query the endstops during the activate_code

I managed to make the BFPTouch "work" by physically adjusting it. However it is a bit too close to the bed and I would prefer to have a control via software.

KevinOConnor, 2019-04-25:
> You need to include a Z movement in your activate g-code.

FYI, it's not valid to place a movement command in a [probe] activate_gcode (nor deactivate_gcode).  See example-extras.cfg for the details.

I'm not sure I understand what the issue is.  The PROBE command is very simple - it descends until the probe triggers.  If you want to run some additional commands on each PROBE attempt, then make a gcode_macro that implements those commands and use that macro instead of directly calling PROBE.

-Kevin

JoaoFelipe, 2019-04-25:
The issue is that the offset configurations (z_offset, x_offset, y_offset) do not seem to make any difference. Maybe I do not understand them. What are they supposed to do?

zwnk, 2019-04-25:
i did the z_offset manually and it works for me. it is possible that you have to change it 0.5 - 1 mm from time to time after removing the bed.
https://www.youtube.com/watch?v=KoCvwPZUkBI

KevinOConnor, 2019-04-28:
@JoaoFelipe - the xyz_offset parameters specify the distance between the tip of the nozzle and the point where the probe contacts the bed.  They don't directly alter movement.  Follow the directions at https://www.klipper3d.org/Probe_Calibrate.html to calculate those three values.  If the probe you are using requires space between the probe and the bed prior to probing then you'll need to make sure that you always move the head away from the bed prior to issuing a probe command (you can do that via a gcode_macro or similar mechanism).

-Kevin

JoaoFelipe, 2019-05-07:
I was misunderstanding the meaning of the offsets. It seems to be working properly

closed 2019-05-07 by JoaoFelipe