Klipper3d/klipper #1546: Unexpected keyword arguments: {'exclusive': True}

Delsian, 2019-04-14:
Looks like commit 7373e9209214cd90cf98498e0f5d94375ddfb115 not compatible with some systems.
---
pi@klippy:~/klipper $ pip freeze |grep serial
pyserial==3.4
pi@klippy:~/klipper $ uname -a
Linux klippy 4.14.79-v7+ #1159 SMP Sun Nov 4 17:50:20 GMT 2018 armv7l GNU/Linux
pi@klippy:~/klipper $ git log
commit 9cb21de68ead14fd0e13ea0f730b79972a82a5fd
Author: Trevor Jones <trevjonez@users.noreply.github.com>
Date:   Sun Apr 14 09:34:22 2019 -0600

    virtual_sdcard: consistent sorting (#1387)

    Signed-off-by: Trevor Jones <trevorjones141@gmail.com>

commit 7373e9209214cd90cf98498e0f5d94375ddfb115
Author: Daniel <daniel.a.joyce@gmail.com>
Date:   Sun Feb 10 08:51:29 2019 -0800

    serialhdl: Enforce exclusive access on POSIX, this is default behavior on windows. (#1216)

    Signed-off-by: Daniel Joyce <daniel.a.joyce@gmail.com>

----
Send: status
Recv: // Internal error during connect: unexpected keyword arguments: {'exclusive': True}
Recv: //
Recv: // Once the underlying issue is corrected, use the "RESTART"
Recv: // command to reload the config and restart the host software.
Recv: // Printer is halted
Recv: // Klipper state: Not ready
Recv: !! Internal error during connect: unexpected keyword arguments: {'exclusive': True}

Delsian, 2019-04-14:
Ok, on my system problem resolved after removing this commit:
Send: status
Recv: // Klipper state: Ready
Recv: ok

agluszyk, 2019-04-14:
Same here happens in my setup (Tronxy x5s) :
`click_pin = ^!PA5
=======================
Extruder max_extrude_ratio=0.266081
Starting serial connect
Unhandled exception during connect
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 137, in _connect
    cb()
  File "/home/pi/klipper/klippy/mcu.py", line 611, in _connect
    self._serial.connect()
  File "/home/pi/klipper/klippy/serialhdl.py", line 61, in connect
    self.serialport, self.baud, timeout=0, exclusive=True)
  File "/home/pi/klippy-env/local/lib/python2.7/site-packages/serial/serialutil.py", line 233, in __init__
    raise ValueError('unexpected keyword arguments: {!r}'.format(kwargs))
ValueError: unexpected keyword arguments: {'exclusive': True}
`

KevinOConnor, 2019-04-14:
Thanks.  I'd appreciate it if both of you could attach the Klipper log file from the event.  (That will help me track down why it passes my tests but fails for others.)

I will revert the change.  @DanielJoyce - FYI.

-Kevin

agluszyk, 2019-04-15:
[klippy.log](https://github.com/KevinOConnor/klipper/files/3080907/klippy.log)
[distro.txt](https://github.com/KevinOConnor/klipper/files/3080911/distro.txt)
I've attached klippy.log and info about OS distro (octoprint image running on rpi3)

KevinOConnor, 2019-04-16:
@agluszyk - can you run `~/klipper/scripts/install-octopi.sh` again and then retry?

-Kevin

agluszyk, 2019-04-18:
after running `install-octopi.sh` it's running fine, so it was just missing upgrade step. Thanks for help!

closed 2019-04-22 by klipper-gitissuebot