Klipper3d/klipper #1551: Timeout during endstop homing

voodoo-bravo, 2019-04-15:
Hello there, 

I'm getting the Error `Failed to home x: Timeout during endstop homing` when issuing a `G28 X0 Y0` on my IDEX printer.

If T0 is near it's endstop homing works just fine. 
As soon as T0 is farther than 10cm away from its endstop, for example after a Print, the carriage moves just a few cm in the direction of the endstop, stops and klippy spits out the above mentioned Error. 

On the other hand, T1 as well as Y can be positioned anywhere and home just fine. 

Example end of Print Gcode:
```
[...]
G1 X109.471 Y100.100 E25.27188
G1 X104.589 Y100.100 E25.41684
G1 X104.571 Y99.811 F16500.000
G1 E23.41684 F2400.00000
G92 E0
M107
; begin custom endcode

G91 ; relative positioning
G1 E-3 F4800
G1 F5000 Z+0.5 X+20 Y+20
T0
G28 X0 Y0
G90 ; absolute positioning

TURN_OFF_HEATERS

; end custom endcode
```
[klippy.log](https://github.com/KevinOConnor/klipper/files/3081389/klippy.log)

I'm currently using de7444ca695b36dd6961055fc7b25d01b7f4c2b4 
Any idea whats going on here?

-Markus

KevinOConnor, 2019-04-21:
Best guess is that your step_distance parameters are not correct.  (They're nominally set to move about 1 micron per micro-step - which is quite unusual.)  Make sure to run through the STEPPER_BUZZ check on all steppers - see: https://www.klipper3d.org/Config_checks.html

-Kevin

voodoo-bravo, 2019-04-21:
The step_distance for all steppers are correct.
I'm using 1/128 Microsteps on 1,8° Motors with 20T GT2 Pulleys on all axis.
Also - printig works just fine and prints are dimensionally accurate.

KevinOConnor, 2019-04-28:
Very strange.  I don't know why that would be.  If you can, reproduce the event and run M112 immediately after the error is produced - see https://www.klipper3d.org/Contact.html#issue-reporting for details.

-Kevin

voodoo-bravo, 2019-04-29:
Here's a [klippy.log](https://github.com/KevinOConnor/klipper/files/3128856/klippy.log) with an `M112` right after the Error, as requested.

KevinOConnor, 2019-05-01:
The log indicates that the initial home of the primary X carriage succeeded, and then on the second home attempt (of the primary X carriage) the endstop did not report a trigger.  Otherwise, the movement commands issued to the stepper/end_stop look to be correct.  Are you using a special type of endstop switch that may be failing?

-Kevin

voodoo-bravo, 2019-05-05:
I had a look at my wiring harness and it seems there was a loose contact in the plug that connects the endstop which caused the unpredictable behavour. 
Sorry for the troubles, that was an oversight on my end. 

-Markus

closed 2019-05-05 by voodoo-bravo