Klipper3d/klipper #1552: Pause before and after travel move

Paulbsh, 2019-04-15:
Hi there,
When a travel move happens, the printhead stops for about a second before and after the travel move.
This sometimes creates blobs.
Any idea about this?
Here's my config: https://pastebin.com/WrgbmTSH

Klippy log attached
[klippy.log](https://github.com/KevinOConnor/klipper/files/3081606/klippy.log)

FHeilmann, 2019-04-16:
This is the resulting load-graph from the config you shared:
[image]
If I assume correctly you are asking for 300mm/s travel moves with a 0.00625 step distance (either via a significant down-gearing or 1/32 microstepping). Either your arduino can't handle the load or your pi is struggling to provide the required steps (either the pi is too slow/old or your SD card is crapping out).

KevinOConnor, 2019-04-21:
FYI, if the rpi or the micro-controller can not keep up then Klipper will report an error.  Klipper will not add pauses due to limited processing power.  (OctoPrint not keeping up can lead to pauses, but neither the graph nor the log indicates a systemic issue there.)

It's hard to say what could be causing the pauses - it's possible there is pauses in the g-code.  (I can't think of anything other than the g-code input that would lead to full second pauses between moves.)

Otherwise, you could try posting the g-code file you are using and post a video showing what it is doing.

-Kevin

KevinOConnor, 2019-05-16:
I'm going to close this issue as it appears the conversation has concluded.

-Kevin

closed 2019-05-16 by KevinOConnor