Klipper3d/klipper #1553: Is it possible to disable the z-axis and extruder?

Uka4me, 2019-04-16:
Google translation, sorry ...
Hello!
Successfully use your product in my printer. For this reason I would like to use klipper for the laser ...
In fact, I have no z axis, only x and y. Kinematics - corexy.
I tried to configure:
```
[stepper_x]
step_pin: ar54
dir_pin: ar55
enable_pin: !ar38
step_distance: .01
endstop_pin: ^ar3
position_endstop: 0
position_max: 200
homing_speed: 50

[stepper_y]
step_pin: ar60
dir_pin: ar61
enable_pin: !ar56
step_distance: .01
endstop_pin: ^ar14
position_endstop: 0
position_max: 200
homing_speed: 50

[stepper_z]
step_pin: ar46
dir_pin: ar48
enable_pin: !ar62
step_distance: .0025
endstop_pin: ^ar18
position_endstop: 0.5
position_max: 200

[mcu]
serial: /dev/ttyACM0
pin_map: arduino

[printer]
kinematics: corexy
max_velocity: 300
max_accel: 3000
max_z_velocity: 25
max_z_accel: 30
```
It is logical, but I get an error saying that there is no z. Adding z:
```
[stepper_z]
```
With an empty description of z, the firmware stops swearing, but when loading gcode, with the G28 command, you give an error timeout switch x.
The laser starts at the command M42 P11 S.
In Repetier firmware, this gcode works properly.

Can I use only axes in klipper: x and y? How to disable the z axis and extruder?

Arduino + RAMPS 1.4 + PC + OctoPrint.
An example of gcode, which I am trying to feed to the firmware:
```
G21
G90
G0 X93.70 Y81.34
M42 P11 
G1 F2
M42 P11 S0.0000
G1 X93.70 Y81.34
M42 P11 S1.0000
G1 X94.00
M42 P11 S2.0000
G1 X95.00
M42 P11 S0.6667
G1 X95.10
M42 P11 S0.0000
G1 X106.70
M42 P11 S0.6667
G1 X106.80
M42 P11 S1.6667
G1 X107.00
M42 P11 S2.0000
G1 X107.40
M42 P11 S1.0000
G1 X107.80
M42 P11 S1.6667
G1 X108.00
M42 P11 S2.6667
G1 X108.20
M42 P11 S3.3333
G1 X108.40
M42 P11 S6.3333
G1 X108.60
M42 P11 S8.3333
G1 X110.20
M42 P11 S8.0000
G1 X110.40
M42 P11 S21.6667
G1 X110.60
M42 P11 S51.6667
G1 X110.80
M42 P11 S96.3333
G1 X111.00
M42 P11 S150.0000
G1 X111.20
M42 P11 S81.6667
G1 X111.40
M42 P11 S7.3333
G1 X111.60
M42 P11 S4.0000
G1 X111.80
M42 P11 S2.0000
G1 X112.20
M42 P11 S1.0000
G1 X112.30
M42 P11 S0.0000
G1 X112.50 Y81.54
M42 P11 S1.0000
G1 X112.20
M42 P11 S4.0000
G1 X112.00
M42 P11 S12.6667
G1 X111.80
M42 P11 S36.6667
G1 X111.60
M42 P11 S77.6667
G1 X111.40
M42 P11 S126.3333
G1 X111.20
M42 P11 S168.6667
G1 X111.00
M42 P11 S147.3333
G1 X110.80
M42 P11 S124.6667
G1 X110.60
M42 P11 S98.6667
G1 X110.40
M42 P11 S85.0000
G1 X110.20
M42 P11 S81.0000
G1 X110.00
M42 P11 S80.3333
G1 X109.80
M42 P11 S81.3333
G1 X109.60
M42 P11 S80.3333
G1 X109.40
M42 P11 S80.6667
G1 X108.60
M42 P11 S58.3333
G1 X108.40
M42 P11 S29.3333
G1 X108.20
M42 P11 S26.3333
G1 X108.00
M42 P11 S22.0000
G1 X107.80
M42 P11 S15.3333
G1 X107.60
M42 P11 S9.0000
.......
```

FHeilmann, 2019-04-16:
I'll go through your issues one by one:
Instead of running `G28` you can use `G28 X Y` to only home the X and Y axii. I believe this solves the first issue you were describing.
Regarding your second issue, you can define your laser pin as an `output_pin` as per [this example](https://github.com/KevinOConnor/klipper/blob/master/config/example-extras.cfg#L735).

You can then imitate the M42 gcode call with the use of a `gcode_macro`. these macros can take parameters, see [here for an example how this was used to implement the M300 gcode in klipper](https://github.com/KevinOConnor/klipper/blob/master/config/sample-macros.cfg#L7). To me it looks like you can simply ignore the `P` part of M42 and always control your laser pin with the given value.

Be advised, however, that other users had some issues 
with using output_pins as laser pins. They reported their issue in https://github.com/KevinOConnor/klipper/issues/133

Hope that helps
-Florian

Uka4me, 2019-04-17:
Thank! `output_pin` helped, as did `G28 X0 Y0`. But unfortunately I deceived you, `[stepper_z]` is not empty ... I filled it with the values from the example:
```
[stepper_z]
step_pin: ar46
dir_pin: ar48
enable_pin: !ar62
step_distance: .0025
endstop_pin: ^ar18
position_endstop: 0.5
position_max: 200
```
If the section is left empty, it swears and requires you to specify a pin, I take the pin from the example without any changes and everything starts working.
I don’t have anything connected to z: neither the stepper motor, nor the limit switch, but it still requires me to specify a pin.
If left as it is, it seems to work. But could the firmware clearly indicate that there is no z axis?

closed 2019-04-24 by klipper-gitissuebot