Klipper3d/klipper #1555: [Feature Request] Add wait time to SET_TEMPERATURE_FAN_TEMP (emulate M191)

fhtagnn, 2019-04-16:
Klipper already implemented several components for chamber heated or cooled system (generic_heater, temperature_fan).  Controlled via extended gcodes is very useful, especially the recently added SET_TEMPERATURE_FAN_TEMP.

However, I would like to emulate M191 (heat up and wait) to have the printer wait until the chamber reaches a certain temperature. Then fan control should kick in an keep it at the set temperature.

This is useful, if you have a chamber without a heater (e.g. Voron 2.1) which heats up due to the bed. With a wait command like M191 heat soak before a print starts can be automated.

I already played around with the code and something like this would implement some form of waiting:

```
if 'WAIT_OFFSET' in params:
   offset = self.gcode.get_float('WAIT_OFFSET', params, 0)
   eventtime = self.reactor.monotonic()
   current_temp, target_temp = self.get_temp(eventtime)
   while (current_temp < (temp - offset)):
     eventtime = self.reactor.pause(eventtime + 1.)
     current_temp, target_temp = self.get_temp(eventtime)
```
This blocks communication, though, and leads to Octoprint loosing communication. I'm not very familiar with the klipper code base, but to my limited understanding gcodes M190 and M109 are realized with a background loop that ensures temperature messages. Do I need to replicate this in the the temperature_fan module? Is this even desirable or is a modification of gcode.py necessary?

KevinOConnor, 2019-04-21:
It's fine if you (or someone else) wishes to implement this.  Yes, it would require sending temperature messages periodically to make sure OctoPrint is happy.  See how the bg_temp() code in gcode.py does this.  My recommendation would be to add a new WAIT_FOR_TEMPERATURE command to heater.py (so that it works with all temperature sensors).

-Kevin

KevinOConnor, 2019-05-16:
I'm going to close this as it appears the conversation has concluded.  If someone wishes to implement it, feel free to submit a github pull request.

-Kevin

closed 2019-05-16 by KevinOConnor