Klipper3d/klipper #1557: how do we make steppers go super fast for a remote direct drive extruder?

brightblack, 2019-04-18:
Hi, apologies for somewhat abusing the github issues tab perhaps, but the documentation & examples haven't really helped me understand how I can get klipper to spin a stepper at ludicrous speed without the same need for microstep accuracy.

I have a flex3d DDA - but it would be similar for users of any flex3d or nimble-type cable-driven direct-drive extruder, where the extruder steppers are geared down at 30:1 or 40:1 - it means I'm not sure if I've hit limits on how fast I can get the extruder stepper to go, but I've been topping out at 25mm/s where ideally I'd like to hit 50mm/s retracts, but think this may be my own lack of understanding on stepper config.

I'm actually using a retro-fitted einsy rambo with trinamic 2130's.  

Current relevant sections are:
`[tmc2130 extruder]
cs_pin: PK4
microsteps: 8
interpolate: True
run_current: .513757
hold_current: .1
#run_current: .613757
sense_resistor: 0.220
diag1_pin: !PK3
driver_IHOLDDELAY: 8
driver_TPOWERDOWN: 0
#driver_BLANK_TIME_SELECT: 2
driver_TOFF: 3
driver_HEND: 1
driver_HSTRT: 5
driver_PWM_FREQ: 2
driver_PWM_GRAD: 4
driver_PWM_AMPL: 240
driver_PWM_AUTOSCALE: True
driver_SGT: -2`

w1ebr, 2019-04-19:
Hello! I have a similar setup. The stepping limit in the system is the processor on the Rambo card. If you can reduce your micro-stepping to 4 (instead of 16, which was what I had and I am making an assumption that this is what you are currently using) you will still have a fine enough resolution for extruding and should be within the stepping limits of the micro-controller.

closed 2019-04-25 by klipper-gitissuebot

elcojacobs, 2019-05-14:
I was able to find this super speed mode for my ender 3. I posted here:

https://github.com/KevinOConnor/klipper/issues/1363#issuecomment-492355285

I'm can do 60mm/s retractions with a 1:3 gear ratio, which would be 180 mm/s on a non-geared extruder.