Klipper3d/klipper #1558: Sudden stop printing

bdn76, 2019-04-18:
RPi 3B+ + MKS Gen-L 1.0 + 3xTMC2208 (XYZ) + 1xDRV8825 (extruder). Print without Octoprint direct from virtual SD (same behaviour with Octo). Printing stop suddenly after 4-5 hour but not always, sometime successfully printing aroud 12-14 hours. I'm check dmesg - no error about MCU disconnect and no error in /var/log/messages on RPi, but MCU is not responding.  
[klipper.log](https://github.com/KevinOConnor/klipper/files/3094303/klipper.log)

KevinOConnor, 2019-04-21:
Nothing in the log indicates a problem (up until the point that the micro-controller stops responding).  Best advice we have is at: https://www.klipper3d.org/FAQ.html#i-keep-getting-random-lost-communication-with-mcu-errors

-Kevin

bdn76, 2019-04-21:
Hmm... I'm tried all of this but without success - print suddenly stop after 8-10 hours. If MCU side klipper code hang would be tty device disconnect from RPi (software tty USB emulation on MCU like STM)? I'm not see tty disconnect in dmesg after stop print.

KevinOConnor, 2019-04-24:
The fact that you're not seeing a USB level reset indicates the USB to serial converter on the board did not reset - but the fault could have occurred at the atmega2560.  It's also possible some other software on the host machine (it looks like you are running a custom install) may have messed up the serial port.  You could try with the latest code to see if the "exclusive serial" change improves things.  (See the FAQ for the upgrade instructions.)

-Kevin

KevinOConnor, 2019-05-16:
I'm going to close this issue as it appears the conversation has concluded.

-Kevin

closed 2019-05-16 by KevinOConnor