Klipper3d/klipper #1562: SCREWS_TILT_CALCULATE inverted rotation indication

claustro, 2019-04-19:
Hi all , maybe I misunderstood something but doing the screws calibration give me inverted advises. For example it say I need to turn a screw CCW instead of CW. Other then it works correctly.

Another question , I am using M4 screws for bed tilt but what about using different screws with different pitch? Is there a way to input a specific  pitch in firmware?
[klippy.zip](https://github.com/KevinOConnor/klipper/files/3097700/klippy.zip)

ArDrift, 2019-04-19:
You can define the type and the direction of the screws in the same section you define their position (`[screws_tilt_adjust]`). It goes like:
```
#screw_thread: CW-M3
#   The type of screw used for bed level, M3, M4 or M5 and the
#   direction of the knob used to level the bed, clockwise decrease
#   counter-clockwise decrease.
#   Accepted values: CW-M3, CCW-M3, CW-M4, CCW-M4, CW-M5, CCW-M5.
#   Default value is CW-M3, most printers use an M3 screw and
#   turning the knob clockwise decrease distance.
```
Which you can find under the [example-extras.cfg](https://github.com/KevinOConnor/klipper/blob/master/config/example-extras.cfg#L295). I hope this helps.

claustro, 2019-04-19:
Thank you for answering me I did't notice that part in config ! :-)

KevinOConnor, 2019-04-28:
I'm going to close this as it looks like the original issue was resolved.

-Kevin

closed 2019-04-28 by KevinOConnor