Klipper3d/klipper #1563: Internal error on command:"G29"

LSergM, 2019-04-19:
[klippy.log](https://github.com/KevinOConnor/klipper/files/3097778/klippy.log)
Hi. Could you help me to figure out with this issue.

JohnEdwa, 2019-04-21:
Klipper doesn't have `G29`, you need to use [the extended G-code commands](https://github.com/KevinOConnor/klipper/blob/master/docs/G-Codes.md).

In this case, most likely `BED_MESH_CALIBRATE`.

KevinOConnor, 2019-04-21:
The G29 was a macro in the config.  The underlying error is a "timeout with mcu" error - best advice we have is in the FAQ at - https://www.klipper3d.org/FAQ.html#i-keep-getting-random-lost-communication-with-mcu-errors .

Also, if this is a new setup, make sure the config checks described at https://www.klipper3d.org/Config_checks.html

-Kevin

LSergM, 2019-04-21:
Thank you for fast feedback.
You can close this issue because when I moved all signals on one board instead of two this issue disappeared.

-Serhii.

closed 2019-04-21 by LSergM