Klipper3d/klipper #1564: Changing Font in LCD Screen

joker25, 2019-04-19:
Hi, is there a way to change the font size on the LCD-Screen?
I looked into the python files and found the font as bitarrays. So i am assuming that you are using a framebuffer to draw the stuff. What kind of program did you use to render the fonts to a bitarray? I would like to implement a smaller font, like on Marlin.

Thanks in advance!

KevinOConnor, 2019-04-21:
It's not currently possible to change the font on the popular HD44780 and ST7920 displays (as the code uses the character generators in those displays).  Though, technically the ST7920 could be modified to use the graphics generator for all on screen content.

The uc1701 (and similar) displays do use graphics for all character generation.  As you indicated, the host python code does render all graphics to an internal framebuffer and then transmits that framebuffer to the micro-controller on these displays.  The source of the font is documented in klippy/extras/display/font8x14.py.  All of the lcd code is in the klippy/extras/display/ directory.

-Kevin

mbgroot, 2019-04-25:
> The source of the font is documented in klippy/extras/display/font8x14.py.

Which program can I edit this font with?

mbgroot, 2019-04-26:
I'm proposing a new font. This is my version of the Terminus font. 

[image]
[font8x14.py.txt](https://github.com/KevinOConnor/klipper/files/3121679/font8x14.py.txt)

KevinOConnor, 2019-04-28:
Interesting.  I wonder if @Arksine has any comments on this.  I do wonder if it would be better to stick with a font similar to the character generator on the st7920 (so the different displays generally look similar).  Also, in order for us to commit, it would need to come as a PR and the origin of the font would need to be documented.

Thanks,
-Kevin

mbgroot, 2019-04-28:
Can you show me an example of a font with st7920? In principle, I can draw almost any font. 
Do I need to make a PR?
How do I confirm that I have made this font?

Arksine, 2019-04-28:
I believe the st7920 uses an internal character set.  Its identical (or very close) to the character set we use for displays that require rendering in software.  IIRC the HD44780 uses a similar character set as well (at 5x8).  I think it is possible to upload custom character sets to either, I haven't looked closely).  I do agree that uniformity across the displays is nice, although not absolutely required.

One approach we could take is to add a "font" configuration option to displays that use the software renderer, with the default being the standard VGA font.   This would avoid the inevitable "I like the old font better" vs "I like the new font" debate if we choose to change it.

mbgroot, 2019-04-28:
Adding a choice of fonts seems to me the best option. As far as I understand, it shouldn't be too hard? 
[image]

joker25, 2019-04-30:
Nice Font, but wouldn't it be better to implement a smaller Font for the Menu? Like for example like http://kt8216.unixcab.org/font5x7/index.html
Actually im struggling to implement this smaller type font, because the char width and length is not the same

Arksine, 2019-04-30:
I don't think there is much point in implementing a smaller font.  The display code subdivides the display into a 16x4 grid, with one character per cell.   A smaller font will not result in more characters on the display.

KevinOConnor, 2019-05-01:
FYI, both the st7920 and hd44780 character generators can only use their builtin fonts.  (Both displays have the ability to define custom characters, but the support is so limited that it is really only useful for defining a handful of custom glyphs.)  For the st7920, we could drive the display entirely in graphics mode if we wanted more flexibility.  (The st7920 character generator uses less bandwidth, but there isn't any indication that we need to conserve bandwidth.)

FWIW, it should be possible to update the code from a 16x4 text grid to an 18x8 grid if someone wants to implement that.

-Kevin

KevinOConnor, 2019-05-25:
I'm going to close this issue as it looks like the conversation has concluded.

-Kevin

closed 2019-05-25 by KevinOConnor

robustini, 2019-12-15:
So isn't it possible to have reasonably smaller fonts on st7920?
They are really huge, and it would be very useful to have more lines in the menus, a slightly finer matrix.

Arksine, 2019-12-15:
As things currently stand it isn't possible.

As Kevin said, an interested developer could implement a larger grid and render the fonts/glyphs in graphics mode.

clowrey, 2020-03-14:
Just want to add that I also think the fonts / menus are really huge on Klipper currently and display about half us much useful information at a time as they could. Of course I don't expect the lead development to spend time on this issue if it does not bother them or they do not use these screens. I don't currently have the time to work on it but would definitely appreciate it if anyone else has the time and feels like working on this.
 
Previously I have used Repetier which I prefer vs the Marlin menu system and layout. 
[image]

This video shows some of the menu Repetier fonts too at 33 seconds:
https://youtu.be/8wQiilALDm4

camatthew88, 2021-01-09:
I agree