Klipper3d/klipper #1565: Add permanent XYZ coordinates on LCD screen

claustro, 2019-04-19:
Would be possible to add permanent coordinates indication on LCD?

Thx
[printer.zip](https://github.com/KevinOConnor/klipper/files/3098148/printer.zip)
[klippy.zip](https://github.com/KevinOConnor/klipper/files/3098149/klippy.zip)

KevinOConnor, 2019-04-21:
I'm not sure what is being requested.  If you are requesting that the display always show the X, Y, Z coordinates instead of the "ready" status message, then it is possible but requires some code changes.  Alternatively, it's also possible to use the menu system to create a custom display layout.  There's some info (and examples) on how to do that in the config/example-menu.cfg file.

-Kevin

claustro, 2019-04-22:
Thank you for answering me , yes this is what I was asking for, I'll check  config/example-menu.cfg  thx!

Lenbok, 2019-04-25:
@claustro Can you post what you come up with here please?

KevinOConnor, 2019-05-16:
I'm going to close this issue as it appears the conversation has concluded.

-Kevin

closed 2019-05-16 by KevinOConnor