Klipper3d/klipper #1566: Error reported as overextrusion but firmware might have tried to move carriage instead of extruder

w1ebr, 2019-04-19:
Hello Kevin,
In this last scenario in the log file, I am trying to print with [extruder1] which is on [dual_carriage]. Just before I get the error message from the firmware, the carriage makes a loud noise (similar to when it is trying to move too fast). Please note that the step rates of the two extruders is not the same and if that is the problem I will fix it ASAP.
Thank you!
Gene
[klippy.log](https://github.com/KevinOConnor/klipper/files/3098916/klippy.log)

KevinOConnor, 2019-04-21:
Nothing in the log indicates the source of the error.  But, please be sure to see issue #1289 - if you are performing an extrude only move immediately after a SET_GCODE_OFFSET command then that can lead to problems.  Work around is to issue a bare "G1" command immediately after the SET_GCODE_OFFSET command.

Otherwise, you could try issuing an M112 after the failing command so that this is more info in the log - see: https://www.klipper3d.org/Contact.html

-Kevin

w1ebr, 2019-04-24:
Issuing a bare G1 is a great workaround. Thanks!

closed 2019-04-24 by w1ebr