Klipper3d/klipper #1567: Mesh behaving strangly

derju, 2019-04-19:
Hello,

i´ve got a strange problem with the bed leveling mesh. When i use a lower number of probing points (for example 7x7) everything works fine, but if i increase it the mesh goes nuts. The attached Picture shows the generated mesh for a 13*13 matrix of probing points the bed visualizer doesnt seem to be the problem since the output from BED_MESH_OUTPUT shows these artefact as well.

[image]

And i know that the mesh in the log file is 15x15, and in the config it´s 13x13. I just checked and changing the config to 15x15 to match the mesh results in the same behaviour.

Hope theres a solution since i would really like to accurately probe my bed with  fine grid.
[klippy.log](https://github.com/KevinOConnor/klipper/files/3098992/klippy.log)

Arksine, 2019-04-19:
Most likely you are getting oscillation from the lagrange interpolation algorithm.  The more points you add the more subject you are to oscillation.

You can choose bicubic interpolation which should behave with the default tension, although when probing such a dense grid you may as well set mesh_x_pps and mesh_y_pps to 0, which doesn't interpolate any extra points between the mesh.

As a side note, there is something strange about the [bed_mesh 1] profile.  The number of points do not match the x and y counts.  This shouldn't be possible unless the profile in printer.cfg was changed.  This is likely going to cause issues if you load that mesh.

derju, 2019-04-19:
Thanks a lot Arksine! Just tried a 15*15 mesh with bicubic and mesh_pps both at 0 and now i´m getting a useful mesh without oscillations 👍

closed 2019-04-19 by derju