Klipper3d/klipper #1570: Trigorilla board and TMC2208 drivers - issue with axes not moving smoothly

dragsterbox, 2019-04-21:
Dear all,

Since I upgraded to TMC2208 (FYSETC v1.2), I am having 2 issues that I cannot fix:

1. I never managed to get my axis moving smoothly. Same for homing
2. Stepper motors direction is reversed and despite I change dir_pin it in the firmware, there is no change

I have tried many different configuration with no success.

I will really appreciate if someone can help on this and/or give me documentation.

Attached, my config file and log.
[printer.txt](https://github.com/KevinOConnor/klipper/files/3100890/printer.txt)

[klippy.log](https://github.com/KevinOConnor/klipper/files/3100888/klippy.log)

StefanMeGit, 2019-05-07:
And what happend? Just closing this does not help other people? U screwed up something?

dragsterbox, 2019-05-07:
Hi Stefan,

Please do not close it. I will update this post with my fix and config file tonight. I am sure that it will help people using Piezo probe system.

dragsterbox, 2019-05-07:
Issue is closed

closed 2019-05-07 by dragsterbox