Klipper3d/klipper #1573: bltouch pin_move_time behavior seems wrong.

markwinger, 2019-04-23:


markwinger, 2019-04-23:
I am new to octoprint and klipper so I may be off in my understanding so let me know if I am misunderstanding something. I have a 3dtouch (bltouch clone) and it works well most of the time.  However I sometimes get an error when homing that says:

Send: G28Recv: // Sending BLTOUCH_DEBUG COMMAND=resetRecv: !! BLTouch failed to verify sensor stateRecv: !! BLTouch failed to verify sensor state.

I tried changing pin_move_time and this seems strange.  I set it 10 seconds and noticed it waits 10 seconds every time, not just for the pin move.  For example when home (successfully) it moves to the xy position,  the deploys the pin, then waits 10 seconds, lowers the z axis until trigger, waits 10 more seconds t hen repeats.  I don't know how the code is supposed to work but it seems like the pin_wait_time should be a timeout, not a wait time.

I assumed the pin_wait_time would be a timeout period where if the pin does not deploy or retract, it gives an error.  But instead it seems to wait until the time period has passed then check if deployed.  Is this correct?  

The klippy log has the pin_wait_time set to 5 and I did a home just before the m112.
[klippy.zip](https://github.com/KevinOConnor/klipper/files/3107708/klippy.zip)

markwinger, 2019-04-23:
Sorry-closed it accidently (first timer).  Reopening...

KevinOConnor, 2019-04-28:
It may be necessary to set `pin_up_reports_not_triggered: False` on a bltouch clone.  See the config/example-extras.cfg file for the details.

The pin_wait_time is not a timeout value, but the absolute time to wait after commanding the pin to move.  It's difficult to implement a timeout because the readings from various clones may not be accurate while the pin is still moving.

-Kevin

markwinger, 2019-04-28:
Are you saying klipper cant read the state of the pin until after the wait time?  I set the time for 10 seconds, the pin moves immediately, but klipper waits 10 seconds after the pin moves.  Is this a problem where you can only check the state once so you have to wait so you dont test too soon?

KevinOConnor, 2019-04-28:
Klipper can read the state at any time.  The reported state may not have any meaning if the pin is in motion, however.

-Kevin

KevinOConnor, 2019-05-16:
I'm going to close this issue as it appears the conversation has concluded.

-Kevin

closed 2019-05-16 by KevinOConnor