Klipper3d/klipper #1574: [question] Access the I²c (Raspi) bus from within the extra plugins?

BKLronin, 2019-04-23:
I´d like to add another sensor that is read non time critical during print. I´m not sure if it would be better to leave this task to the mcu and call it from within klipper or directly access the RPI i²c. 

Appreciate any advice on this. :)

theopensourcerer, 2019-04-23:
Just maybe, have you seen this plugin for Octoprint?

https://github.com/vitormhenrique/OctoPrint-Enclosure

I am using it with a 1-wire temp sensor but it does support other interfaces too I believe. The sensors are plugged into the RPi rather than the MCU. Just a thought.

BKLronin, 2019-04-23:
Thanks, thats interesting, I missed that. Considered to implement my stuff into Octoprint at first but wasn´t sure if its better to have it on  a lower level.

KevinOConnor, 2019-04-24:
If you do wish to access i2c from the rpi, then my recommendation would be to run the Klipper Linux mcu code on the rpi and use that to pass along the i2c messages.  (It would require some additional C code, but most of that is in place already with the pca9685.c code.)

So, in a nutshell, my recommendation is to use the micro-controller code to talk with peripherals, and to restrict the host code to communication with the micro-controller code.

-Kevin

closed 2019-04-30 by BKLronin