Klipper3d/klipper #1575: How to do "Extruder linearity correction calibration" on TMC2130?

Elocai, 2019-04-23:
https://help.prusa3d.com/article/p2xdpg0ul8-extruder-linearity-correction-calibration

I have this moire effect on my TMC2130s and can´t find the commands or values in klipper that relate to 
TMC_SET_WAVE_E0
TMC_SET_STEP_E0
to fix it

is this just one the tmc variables or am I missing something?

Elocai, 2019-04-23:
[not an use issue.log](https://github.com/KevinOConnor/klipper/files/3109065/not.an.use.issue.log)
here bot

Elocai, 2019-04-23:
seems like it uses the wave table on page 78 of the manual and applies a power function to it and the corresponding registers are on page 30

FHeilmann, 2019-04-24:
This is a feature implemented in the Prusa3D marlin fork. Klipper has no such functionality (yet). That's the reason you can't find any corresponding commands.

Elocai, 2019-04-24:
could someone implement it or is anyone working on it or interested in doing so?

KevinOConnor, 2019-04-25:
It's my understanding that @Arksine has done some work on this.  I don't know the state of it, however.

In the future, please do provide the Klipper log file.

-Kevin

megablue, 2019-04-28:
according to this article, correction table will actually make the issues worse....

https://www.micromo.com/media/wysiwyg/Technical-library/Stepper/6_Microstepping%20WP.pdf 

i dont know enough to comment on these...

Arksine, 2019-04-28:
I did port Prusa's linearity correction to the PrusaOwners fork, however I haven't tested it against the recent changes to tmc2130.py, so I'm not sure if it still works.  @megablue is generally correct, in most cases linearity correction isn't helpful unless you have the proper equipment to tune your motors.

Prusa's implementation applies specifically to their LDO motors and the stock extruder.  It scales the wave a bit which does diminish moire.  It doesn't work for other motors, and it doesn't do much for geared extruders.

Elocai, 2019-05-04:
Maybe I was a bit off, the issue I thought that would be moire artifacts was actually fine direction shift on the model - "details" or slicer artifacts you could call it.

Now I would like to use sensorless homing but can´t find how to set vsense for the printer.cfg, probably will open up a new feature request for that.

jiri-jirus, 2019-05-07:
@Arksine sorry, but there is nothing correct on @megablue opinion that correction table will make thins worse. extruder stepper is specific because it's pushing against big resistance, so lack of torque is very visible as periodical lack of extrusion force followed by miniature overextrusion. very visible on 20mm calibration cube. 
LDO motors ale nothing specific, their parameters are known. inductance of 13mH - affect max speed but not relatively slow extruder

Arksine, 2019-05-07:
@jiri-jirus If you do not tune the wave correctly then changing wave will indeed make extrusion worse.  You need a scope and preferably a dev-kit to do it, there is no one size fits all way of tuning motors procedurally.

As I stated, the algorithm ported from Prusa Firmware simply scales the default wave.  This works okay on the specific LDO extruder motors that ship with the MK3, it visibly reduces moire.  Though my own testing and feedback from others, scaling the wave does not help with motors of a different spec, and in fact makes things worse.

KevinOConnor, 2019-05-25:
I'm going to close this issue as it looks like the conversation has concluded.

-Kevin

closed 2019-05-25 by KevinOConnor

Nandox7, 2021-01-28:
@Arksine where I find the implementation you made?
I have that problem with my TMC5160. Tried two steppers so far and no luck and Id like to keep klipper as a FW.