Klipper3d/klipper #1576: Probing problem with Piezo probe and delta printer

dragsterbox, 2019-04-23:
Hi Everyone,

I am having a problem when probing on my Anycubic kossel linear plus. When the nozzle touches the bed, the piezo probe triggers but does not stop moving immediately. As a result, z height is too low and the nozzle is touching and putting pressure on the bed. 

Until now, I did not manage to sort this problem out. I am also wondering if possible to do a multiple touch (average calculation). 

Thanks for your help

Link to video: https://www.facebook.com/carabottav/videos/10157346569503628/?t=2

[klippy.log](https://github.com/KevinOConnor/klipper/files/3109275/klippy.log)

[printer.txt](https://github.com/KevinOConnor/klipper/files/3109258/printer.txt)

ReprapRyn, 2019-04-23:
I don't see a mesh bed leveling or tilt section in your  config. I also don't see anything with how you are calibrating your z probe.

how much deform do you have in your piezo?

Hywelmartin, 2019-04-24:
piezo probes need some speed to generate the trigger voltage. then it will trigger sooner. are you sure that it's connected right, triggers when pushed and not when you release the pressure. 
my z_offset: -0.12

dragsterbox, 2019-04-25:
Hi ReprapRyn,

I wanted to fix this basic probing issue first before working on mesh levelling and/or macros. Another user pointed out that may be the mount is too stiffly and that's the reason why it is hitting the bed. 

I am currently working on a new design that must fix this stiffness issue. Must be ready by the end of the week.

Hi, Hywelmartin,

May be I must increase probing speed as you suggested. I confirm that the probe is connected right and triggering only when there is pressure on the nozzle.

Hywelmartin, 2019-04-25:
there is a lot of information in this series of YT clips from precision piezo https://youtu.be/WDHU3tRU-C4?t=167

KevinOConnor, 2019-04-28:
@dragsterbox - the log you attached just shows that the firmware needs to be recompiled during an upgrade (see the FAQ for the full upgrade instructions).  Without a more meaningful log, there's little we can do to assist.

-Kevin

dragsterbox, 2019-04-29:
Thanks Kevin for pointing it out. Here is the new log. 
[klippy.log](https://github.com/KevinOConnor/klipper/files/3128178/klippy.log)

KevinOConnor, 2019-05-02:
That second log is identical to the log originally posted.

-Kevin

dragsterbox, 2019-05-02:
Please let me know if this one is alright. 

In the mean time, I designed a full piezo system with spring system but I didn't notice any improvement when probing. I suppose that this issue is software related.

[image]

[klippy.log](https://github.com/KevinOConnor/klipper/files/3139708/klippy.log)

KevinOConnor, 2019-05-09:
Nothing in the log looks suspicious.  If you're unsure, you can run the problematic command (eg, PROBE) and then after the problem immediately issue an M112 command (as described at https://www.klipper3d.org/Contact.html ).  Then provide that klipper log file (which will contain more information).

That said, Klipper moves the motors until it gets a signal from the probe.  It's the same code that works on every kind of printer with every type of probe.

-Kevin

closed 2019-05-25 by KevinOConnor