Klipper3d/klipper #1577: Setting a long G4 (dwell) time causes printer to timeout & disconnect

brightblack, 2019-04-24:
If you send a high value to pause the printer (say because you want to use the bed to heat an enclosure), it causes the printer to go unresponsive (guess the arduino is busy-waiting & not responding to serial during a G4, which eventually makes klipper d/c)
Send: G4 P60000
Recv: ok
No response from printer after 3 consecutive communication timeouts, considering it dead. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Changing monitoring state from "Operational" to "Offline (Error: Too many consecutive timeouts, printer still connected and alive?)"
Connection closed, closing down monitor
[klippy.log](https://github.com/KevinOConnor/klipper/files/3112696/klippy.log)

megablue, 2019-04-24:
this has nothing to do with klipper as far as i know. this is the timeout detection implemented by octoprint, you can disable it or set a lenient timeout period.

brightblack, 2019-04-25:
hmm.. so you're right - changing the timeouts settings can make a difference - by making them huge i succeeded in having it take over half an hour to timeout - but i'm attempting to use my enclosure to dry out filament for 2hrs - and it still ends up timing out - I suspect making octoprints timeout connection values very large it's really just a bandaid masking a deeper issue that the temperature-update keep-alives should ideally still happen during G4 commands I think, where I suspect it makes them not; through a combination of very long keep alives and running batches of shorter G4 commands maybe i can workaround this - but this is a poor user experience for people trying to use klipper that seems unnecessary if it's possible to get the host to keep responding during G4's - which is I think what marlin or other firmwares do ?

KevinOConnor, 2019-04-28:
This is the behaviour of the G4 command.  In general, g-code is terrible - there isn't much that can be done to "improve" it.

-Kevin

brightblack, 2019-05-06:
drat - thanks for taking a look then anyway i guess

closed 2019-05-06 by brightblack