Klipper3d/klipper #1579: Delta Calibration Question

novio8, 2019-04-25:
Hi, this is just a question regardin Delta Calibration.
First I did Basic delta calibration and then Enhanced delta calibration.
Printer works, but I had to change the Glass printing Surface.
To adopt to new surface I'v run Basic delta calibration again without
running Enhanced delta calibration. 
Is this OK or not ?
Regards
MArio

mjf55, 2019-04-27:
I would suggest you run it again.  When I compare my printer.cfg before and after the advanced calibrate process, I see that there are changes to delta radius, angles, arm_length and position_endstops.  Also, and probably most important, it the whole section starting with distance0 .  If you still have those, they would be incorrect because of the glass.
If it were me, I would re-run the enhanced calibration process.

novio8, 2019-04-27:
mjf55 thanks, that what I thought

KevinOConnor, 2019-04-28:
The measurements taken from an extended calibration are saved and then later used during future basic calibrations.  If you make a significant change to the printer then it would make sense to rerun the extended calibration.

-Kevin

closed 2019-04-28 by novio8