Klipper3d/klipper #1580: Multiple extruders with one heater on LCD display

dystechnic, 2019-04-25:
Hi,

My printer has three extruders wich use one hotend. So I configured Extruder 1 and Extruder 2 to use the heater of Extruder. (shared_heater: extruder)
My display shows two nozzle icons. The temperature of both nozzles is the same wich makes sense because there is only one heater.

In the display.py file I noticed that there is a check to see if Extruder 1 is configured. self.extruder1 = self.printer.lookup_object('extruder1', None) If I am not mistaken.
In my setup this returns ' True' so a second nozzle icon is displayed on the LCD. 
Because I use only one heater, the second icon does not make sense. 

Would it be possible to do a check if a shared heater is used when multiple extruders are defined in the config file? If that's the case then only one nozzle icon should be displayed wich makes more sense to me.

[klippy.log](https://github.com/KevinOConnor/klipper/files/3117115/klippy.log)

KevinOConnor, 2019-04-28:
It's possible to implement something like the above - if someone wishes to do that and submit it then that's fine.  FWIW, you may have better luck using the menu system to make a customizable display - see config/example-menu.cfg for details.

-Kevin

dystechnic, 2019-04-28:
Thank you. 
I will have a look at the menu.cfg

closed 2019-04-28 by dystechnic