Klipper3d/klipper #1581: BLTouch on 2nd MCU

laurencefourie, 2019-04-25:
Good day Kevin. I have an Ender 3 Pro and as you probably know there are limited connection pins on the stock board. So I was wondering if it would work if I added another mcu - Arduino Nano- and connected and configured the BLTouch on the extra mcu?

laurencefourie, 2019-04-25:
My apologies. I came across another thread that explains that  the Z Stepper need to run on the same mcu as the BLTouch. Could one move the Z stepper to the 2nd mcu and then setup the BLTouch on the 2nd mcu?

jakep82, 2019-04-25:
https://github.com/KevinOConnor/klipper/blob/master/config/example-multi-mcu.cfg

laurencefourie, 2019-04-25:
Thank you Jake. So do you think that this is a viable solution?

jakep82, 2019-04-25:
Yes, many people run the Z axis on a separate MCU, and use a probe attached to that MCU.  See the Voron for an example.

laurencefourie, 2019-04-26:
Thank you for your assistance.

closed 2019-04-26 by laurencefourie