Klipper3d/klipper #1583: Firmware constant 'PWM_MAX' not found

cdtsuipo, 2019-04-27:
[klippy.log](https://github.com/KevinOConnor/klipper/files/3123682/klippy.log)
hi,
    i use a smoothieboard, and i want pin P2.3 be a hardware pwm pin,the system show this message.this pin normally was E0 step pin, so i am sure this pin was hardware PWM capable. also it's working in software mode.is it a config problem ?

THANKS!

KevinOConnor, 2019-04-28:
Klipper doesn't support hardware pwm on the lpc176x chips.  In Klipper, we generally only use hardware pwm with boards that emulate a voltage (via an RC circuit) - typically for stepper current control.  That hasn't been needed on any lpc176x board.  Depending on your setup, the standard software pwm may be sufficient.

-Kevin

cdtsuipo, 2019-04-28:
@KevinOConnor OK ，i got it ,thanks for reply !

closed 2019-04-28 by cdtsuipo