Klipper3d/klipper #1584: Does/Can Klipper use a system's "Real Time Clock" for MCU sync?

jalanjarosz, 2019-04-27:
This is more of a general question, than an issue - 

It's posted: 
> "Klipper supports printers with multiple micro-controllers. For example, one micro-controller could be used to control an extruder, while another controls the printer's heaters, while a third controls the rest of the printer. The Klipper host software implements clock synchronization to account for clock drift between micro-controllers. No special code is needed to enable multiple micro-controllers - it just requires a few extra lines in the config file."

On systems using a SBC, like Raspberry Pi, for controlling Klipper, does Klipper take advantage of a system RTC (Real Time Clock), if available, for keeping multiple MCU's synchronized and help with time sensitive calculations?

Simply put, is it worthwhile to spend the $6 to add a RTC chip to a Pi on a multi-MCU printer?

Thanks, in advance, for your input.

jalanjarosz, 2019-04-27:
[klippy.log](https://github.com/KevinOConnor/klipper/files/3123849/klippy.log)

KevinOConnor, 2019-04-28:
No - Klipper doesn't make use of any special timing hardware on the host machine.  (In general, the Klipper host software is just a regular process without any special hardware access or permissions.)  The Klipper clock synchronization doesn't depend on an absolute time measurement - instead it calculates the clock drift between the micro-controllers.  As such, no special host clock is needed.

-Kevin

jalanjarosz, 2019-04-28:
Kevin, thanks again for your reply.  It's much appreciated, and thank you for all that you have done and continue to do.

closed 2019-04-28 by jalanjarosz