Klipper3d/klipper #1585: Endstop phase correction with wrong sign

benhurstein, 2019-04-27:
Hi.

I was configuring sensorless homing on a delta printer with trinamic 2130 drivers.
The stall detection is not always on the same spot in my printer, sometimes it detects
the stall one full step (16 microsteps) before or after the point where I have calibrated the printer.
To correct this, I activated the endstop phase correction. The problem got worse.

The problem is in endstop_phase.py:get_homed_offset(). It is calculating the offset with the 
opposite sign than what is needed in my printer. The code works OK when the stepper phase 
augments as the motor moves towards the endstop, but in my printer, the phase augments as
the motor moves away from the endstop.

I inverted the sign of the return value of that function and now the printer is working very well.
I do not know how to identify the direction of the phase, to correct this so that it works for both 
cases.

In rare occasions, the stall is recognized two full steps away from the right spot, and in that case 
klipper issues an "incorrect phase" error. Is it possible to detect this problem automatically (in 
octoprint, for example) and re-issue a G28 command? I've never seen it happen twice in a row.

Thank you,
Benhur

[klippy.log.gz](https://github.com/KevinOConnor/klipper/files/3124278/klippy.log.gz)

megablue, 2019-04-27:
tldr; don't use sensorless homing on a delta.

sensorless homing isn't a good idea in the first place if you want a highly repeatable homing more so on a delta due to the nature of trigonometry math.

benhurstein, 2019-04-28:
I did some 50 homings today, to test the repeatability. 
There are 3 motors, so that was 150 stall detections.
About two thirds of them were on the configured phase, all the others were on one of the neighbor phases, which means that they can be automatically corrected with "phase correction".
I'll do some more testings, but it does not look so bad.

KevinOConnor, 2019-04-28:
Interesting.  It's been theorized that one could mix sensorless homing with endstop phase detection, but I think you are the first to report the results of an experiment with it.  It would help a lot if you could provide the exact changes you made (run `git diff` and post the results).

> In rare occasions, the stall is recognized two full steps away from the right spot, and in that case
klipper issues an "incorrect phase" error. Is it possible to detect this problem automatically (in
octoprint, for example) and re-issue a G28 command? I've never seen it happen twice in a row.

I suspect one could update the code to do that, but it would require code changes.

-Kevin

benhurstein, 2019-04-28:
It's just a minus in one line: on line 83 of endstop_phase.py, instead of
  return delta * self.step_dist
I've put
  return -delta * self.step_dist

It works on my printer, because when the carriage moves up towards the endstop, the phase goes down (the phase returned by tmc2130.py:get_phase). You probably tested on a printer where the phase goes up while the carriage is moving towards the endstop. I do not know how to test the direction of the phase, maybe it is given by the inversion (or not) of the DIR pin?

Benhur

KevinOConnor, 2019-04-28:
Yes - this looks like a bug in Klipper.  Klipper is expecting the phase reported by the stepper driver to increase when the axis position increases.  It's not properly taking into account the inversion of the dir pin.  Oddly, the tmc drivers increment when the dir pin is inverted (my test delta printer with the tmc2208 inverts the dir pin).

-Kevin

KevinOConnor, 2019-06-07:
FYI, I think #1710 should address this.

-Kevin

benhurstein, 2019-06-07:
Thanks. I'll be away from the printer for a week, I'll test it when I'm back.

Benhur

KevinOConnor, 2019-06-22:
FYI, I merged the fix into the master branch (commit 80194a7e).  Let us know if you continue to see problems.

-Kevin

closed 2019-06-28 by KevinOConnor

benhurstein, 2019-07-02:
Sorry for replying so late. I can confirm that your change fixes the issue on my printer.
Thank you,
Benhur